sparkfun/OpenLog_Artemis #100: Feature request - programmable "wake up" to trigger start of data logging

chrocket, 2021-09-03:
It would be nice to have a programmable "wake up" mode that would trigger the start of data logging.

For example, an accelerator signal for a certain number of bins on the z-axis would trigger the logger to start saving data to the serial port or SD card.  

Likewise, there might need to be a corresponding "idle" trigger that would turn off data logging

This could be taken from the 6DOF chip or coded explicitly in the OpenLog FW.

I noticed you added support for ICM-20948 and this has a "wake up on motion interrupt" that might be leveraged. THe interrupt pin might require another HW PCB modification or perhaps this could be done in the FW.

chrocket, 2021-09-08:
Didn't mean to close this.

PaulZC, 2021-09-09:
Hi @chrocket ,

If I understand you correctly, you would like us to add a feature to:
- Configure the on-board ICM-20948 for wake-on-motion
- Have the OLA go into a low power state where logging is stopped but the ICM-20948 is still powered
- When the ICM-20948 generates an interrupt, the OLA wakes up and starts logging
- The OLA returns to the low power state after a pre-defined interval of no motion (as measured by the accelerometers)

This is quite a big request. It requires new code for the ICM-20948, a new low power state with wake-on-interrupt, and code to monitor the accelerometers. But it is a reasonable request. In terms of priority, adding basic BLE functionality needs to come first (adding that has been on the things-to-do list for a very long time). So it may be weeks or months before we can look at this.

Best wishes,
Paul

chrocket, 2021-09-09:
Paul, yes this summarizes the request. Thanks. I get it that BLE is very high priority.

Applications are for things like liftoff detection of model rockets, drones, rc planes, or package movement detection.

One simplifying aspect to consider is to have  log data for a specified time interval and then stop logging and go back into low power mode, or have an option for it to wake up and log indefinitely.

Thanks, Curtis

waymond91, 2021-09-14:
I know this isn't strictly relevant to this thread - but when can we expect streaming log data over bluetooth?