sparkfun/OpenLog_Artemis #102: IMU Offline after firmware update

waymond91, 2021-09-13:
### IMU Offline after firmware update
After receiving my OpenLog Artemis everything worked great out of the box. However, I wanted to use the DMP and get quaternion outputs logged, so I updated the firmware to version 20. Once updated, no IMU data is output whatsoever, and when the board starts it outputs "IMU Offline."
I reset the board to its defaults, and I still get the following output at startup (with and without an SD card).
[image]

### Your workbench
* I am using an 8gb sd card formatted with the sdcard tool specified in the docs.
* Everything has been reset to default, so 10hz no DMP.
* I am using firmware version 20
* OpenLog is not connected to any external devices
* The device is powered over USB
* Windows 10

### Steps to reproduce
Plugging in the device and opening a terminal yields no IMU output.
I have updated the firmware multiple times in case the flash went wrong, and even tried with v19 and got the same resut

PaulZC, 2021-09-13:
Hi Brett (@waymond91 ),
Thank you for raising this issue.
Can you please check you have used the correct version of the firmware:
If you have the Red (SparkFun) version of the OpenLog Artemis, you need to use the **OpenLog_Artemis-V10-v20.bin** binary.
The **OpenLog_Artemis-X04-v20.bin** binary is for the older Black (SparkX) version of the OLA.
Best wishes,
Paul

waymond91, 2021-09-14:
That did the trick, thank you!

closed 2021-09-14 by waymond91