sparkfun/OpenLog_Artemis #104: IMU fails to init if SD card is disabled.

harri192, 2021-09-18:
### Subject of the issue
Encountering very weird behavior if I set the settings.enableSD to false.

If I leave a microSD card in after bootloading, then the OLA will continuously reset itself.
If I then pull the microSD card out, the OLA is able to boot up, however I get errors that the IMU failed to initialize. 

### Your workbench
* Are you using a microSD card? If so, what size? How is it formatted (FAT, FAT32, or exFat)?
     Using the standard microSD card that was shipped with the OLA from SparkFun.  Its being formatted with the default formatting option in the OLA.
* What version of firmware are you using? This is shown at boot, for example: "Artemis OpenLog v1.10"
     v2.0
* How is OpenLog Artemis wired to your sensor(s)?
    No external sensors.  Just the OLA board hooked up to my PC over USBC.
* How is everything being powered?
    USBC.  No battery is connected nor is power being supplied via the breakout pins.
* Are there any additional details that may help us help you?
     Found it interesting that in the microSDPowerOn function that the PIN_MICROSD_POWER is being set to LOW and vice-versa for the microSDPowerOff.  This is opposite from what is specified in lets say the imuPowerOn function.

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue.
Try setting the settings.enableSD to false and observe behavior.

### Expected behavior
Tell us what should happen
I expected that if you disable the SD card that the remaining onboard sensors should still operate fine.  Along those same line, even with the enableSD set to false, leaving in an SD card shouldn't cause issues.

### Actual behavior
Changing setting.enableSD to false prevented the IMU for initializing.  If an SD card was left in, the entire OLA would just continously reset itself (which almost made me think I bricked my board because I couldn't re-upload code in that configuration).  If the SD card was removed the OLA was able to run through the whole setup routine but the IMU failed to initialize.

PaulZC, 2021-09-19:
Hi @harri192 ,
Thank you for reporting this issue.
I guess you must be editing the OLA_settings file manually and setting enableSD to false / 0?
Looking at the code, I think enableSD is probably obsolete and has been superseded by logData / logSerial and online.microSD. There appears to be no way to disable enableSD through the menus. It should always be enabled, unless you edit the settings file.
I think the fix will be to remove enableSD from the code - after some careful checking. I will include this in the next release.
Best wishes,
Paul

PaulZC, 2021-12-14:
Thank you for reporting this @harri192 - it is corrected in v2.1.
Best wishes,
Paul

closed 2021-12-14 by PaulZC