sparkfun/OpenLog_Artemis #105: Software reset

alino93, 2021-09-29:
Hi @PaulZC,
I have been using the Slowlogging feature to reduce the power consumption of the boards at night and normal (fast) logging during the day. The boards are powered by a Lipo battery.

My problem is that if the battery dies, OLA does not turn on again automatically even after being charged by a wall adapter. The only way to turn it on again is the reset button or connecting to a computer. I give these sensors to very old people and it would be a lot easier to have a software solution to turn the board on again. Could we have a software solution to reset OLA? Like after fully charged? or at certain times of the day?

Thank you,
Ali

PaulZC, 2021-09-30:
Hello Ali (@alino93 ),

Unfortunately, the feature you are asking for is difficult to implement.

When the main battery dies, the Artemis goes into a very low power state which keeps the Real Time Clock (RTC) running. The only way to wake it from that state is with a reset.
- Generating a reset when the battery is charging or fully charged would require a hardware change (a new circuit board).
- Having the OLA wake up at a certain time of day, or after a period of time, is possible **but** if it does that, and the battery has not been recharged, then it will reset completely and the RTC will reset too.

The only safe way to keep the RTC running is to leave things as they are now.

I am happy to keep discussing this, but I do not think it is possible to give you the feature you need.

I think the simplest solution might be to ask your patients to make sure they recharge the OLA every night?

Best wishes,
Paul

alino93, 2021-09-30:
Hello Paul (@PaulZC), 

Thank you for the clarification. 

I ask the patients to charge the OLA every night but unfortunately, they usually forget.

Ok, a scheduled reset is not going to work because the RTC will rest and there is no easy way to generate reset when recharged. 

Is there anything that I could add to the charger to reset the OLA every time they connect it to? (like when we connect the OLA to a computer)

Regards,
Ali

PaulZC, 2021-09-30:
Hi Ali,

Humm... A standard USB-C charger would not generate the reset. But maybe we could design a very simple board which you could plug in-line to generate the reset after a few seconds...
I shall think more about this!

Best wishes,
Paul

alino93, 2021-09-30:
Paul,

Thank you very much, that would be great!

Best, 
Ali

alino93, 2021-11-19:
Hi Paul (@PaulZC)
I, unfortunately, lost two patients this past month because of this issue. Could you please help me to solve this issue or tell me to look where?
Thank you again,
Ali

vicgwee, 2021-11-19:
Hi Ali, 

I'm really sorry to hear that. 

If I'm not wrong, this board that Paul suggests would need to act as an USB Host, so it can connect with the CH340E chip to tell the OLA to reset. I'm not sure whether anything like that exists on the market right now. 

Just wondering, would it be possible to tell the patients to press the reset button when they charge it? That could be a stopgap solution.

Victor

PaulZC, 2021-11-19:
Hello Ali,

I am very sorry to hear about your patients.

We have not forgotten about your request, but we are a small team and we are working on many other projects at the moment.

As Victor (@vicgwee) says, you need something which can: deliver power to the OLA via USB; and act as a "USB Host". The "USB Host" emulates a computer, and would need to open a serial connection to cause the OLA to reset.

SparkFun do offer a [USB Host Shield](https://www.sparkfun.com/products/9947) and that may do what is required. It would need to be connected to a "[RedBoard](https://www.sparkfun.com/redboards)". You would also need a standard power supply or wall adapter (with a 5.5/2.1mm power jack) to connect to the RedBoard. I do not have one of these boards, but I will try and get one to test.

There are other possibilities, such as this small board from Hobby Electronics: https://www.hobbytronics.co.uk/usb-host-board-v24 . Again, I do not have one of these boards, but I will order one to test. It may work on its own, or it may have to be connected to a microcontroller to open the serial connection. Again, it will need a separate power supply.

If either of these is successful, then - with some design effort - it would be possible to make a compact version dedicated to your project.

I will let you know when I have more news.

Very best wishes,
Paul

PaulZC, 2021-11-19:
Hello Ali,

I have ordered both boards. I will let you know when I have more news.

Best wishes,
Paul

alino93, 2021-11-19:
Hi Paul,

Thank you very much, that would be great please keep me updated. As Victor suggested I have tried to add a reset button and asked the patients to press it every morning but they tend to forget, also pressing it when the battery is empty resets the RTC.

Best,
Ali

PaulZC, 2021-11-21:
Hello Ali,

I have tested both the Hobby Electronics board and the SparkFun USB Host Shield.

The Hobby Electronics board is a very nice little product, but it can not (currently) generate the RTS reset pulse needed to wake the OLA. I have reached out to the Hobby Electronics team to see if it would be possible to add this feature.

I have had limited success with the SparkFun USB Host Shield. I was able to adapt some [code]( https://forum.arduino.cc/t/arduino-usb-host-shield-2-0-ch340-driver-problem/620548) which someone else has written to support the CH340 (the USB interface chip on the OLA):

- This works:
  - OLA is on and logging
  - Shield + RedBoard assembly is on and waiting
  - Connect the OLA to the shield assembly
  - The shield code recognizes and connects to the OLA
  - The shield code is able to reset the OLA (using the RTS signal)
  - Disconnect the OLA
  - Discharge the OLA battery so the Artemis goes into deep sleep
  - Re-connect the OLA to the shield
  - The shield code re-recognizes the OLA, and resets it to wake it up (without losing the RTC time)
  - The OLA continues to charge while logging normally

- This does not work:
  - OLA is in the low battery / deep sleep state
  - The shield assembly is powered off, then on
  - Connect the OLA (still in deep sleep) to the shield assembly
  - The shield assembly does not recognize the OLA and is not able to reset it to wake it from deep sleep
 
- This also does not work:
  - OLA is in the low battery / deep sleep state
  - The shield assembly is powered off
  - Connect the OLA (still in deep sleep) to the shield assembly
  - Power on the shield assembly
  - The shield assembly does not recognize the OLA and is not able to reset it to wake it from deep sleep

So, right now, I do not have a fully-working solution for you.

I know you want to keep your OLA assembly as compact as possible, but would it be acceptable for you to include a [Real Time Clock module](https://www.sparkfun.com/products/16281)? I could alter the OLA code so it would:
- Allow you to set the RTC module time via the OLA serial menus
- Disable the deep sleep functionality - so that the OLA will reset and its code will re-start normally as soon as USB power is connected
- When the code re-starts - the Artemis will automatically set its own RTC from the RTC module and then continue logging normally

Please let me know if that would be an acceptable solution for you.

Best wishes,
Paul

alino93, 2021-11-21:
Hi Paul,

Thank you for testing and working on a solution. So it seems the USB host shield is not working for me because the Redboard should be kept on all the time. 
What about a raspberry pi board, could we program it to open a serial port and reset the OLA? 

About the RTC module, should I add this module to the OLA assembly? Or could it be outside and ask the patients to connect the Qwiic cable to do its job?

Best,
Ali

PaulZC, 2021-11-21:
Hello Ali,

Raspberry Pi is another possibility. It would need (e.g.) a Python script which: runs on start-up; keeps checking for a CH340 connection on USB; if it sees one, toggle the RTS pin to reset the Artemis. It should all be feasible / possible but I have not written that type of Python code before.

The RTC module would be permanently connected to the OLA via a short Qwiic cable. It would not affect the battery life - it’s current draw is very low. But it will of course make each logger larger.

If your patients cannot remember to press the reset button, I do not think they would be able to remember to connect a Qwiic cable during charging?

Best wishes,
Paul

PaulZC, 2021-11-21:
There are some good suggestions here on how to detect if a serial device is connected / still connected:

https://stackoverflow.com/a/49450813

alino93, 2021-11-21:
Paul,

Yes connecting the Qwiic cable is not a good solution and pressing a button seems better. But the assembly is already large and I do not want to make it even larger. I think this way the thickness of my assembly would increase at least 1.5 times which is too bulky to mount on the body (I use 5 sensors on each person).

I will do some research about the Raspberry Pi solution and let you know.

Thank you,
Ali

PaulZC, 2021-11-21:
pySerial includes a function called **rts** which can be used to set the state of the RTS pin.

And here are suggestions on how to run a script on start-up:
https://www.tomshardware.com/uk/how-to/run-script-at-boot-raspberry-pi

Best wishes,
Paul

alino93, 2021-11-21:
Thanks Paul, this helps.
Best,
Ali

alino93, 2021-12-28:
Hi Paul,
I managed to solve the issue a while ago using a Raspberry Pi by installing the CH340 drivers on it and creating a autostart for a small script.
I have explained what I did here for the record: [https://github.com/alino93/OLAAutomaticReset]

I was thinking maybe I can use the Raspberry pi to set the RTC every time it is connected to the OLA (update with the Pi's time). Or other things like downloading the data, etc.  Do you think is that possible?

I think we can close this issue.
Thanks,
Ali

closed 2021-12-28 by alino93