sparkfun/OpenLog_Artemis #111: Occasional erratic sample times at 6 minute sample rate.

ryanneve, 2022-01-11:
### Subject of the issue
Occasional sample timing errors when logging slowly.

### Your workbench
Firmware 2.1
Connected to MS5837-BA02 pressure sensor
Using Qwiic bus to power a BLE module connected to RX & TX.

### Steps to reproduce
Set to log every 360 seconds
No slow log on pin 11
Powered from battery
Logging to serial & SD
Minimum awake time 15000 ms
Minimum Qwiic bus power up delay 10000 ms

### Expected behavior
Samples should be every 6 minutes.

### Actual behavior
Outlier samples below marked with **<-(x)** where x is the effective sample time in seconds when it's not 360.
The length of the erratic sample times don't seem to have a pattern.
This could either be a problem with sleeping for the wrong about of time (sysTicksToSleep), or being woken unexpectedly.

### Example log:
2022-01-06 06:40:02.384313 01/06/2022,06:40:26.26,5.22,-911.13,-18.07,-424.32,24.82,1018.96,21.49,157,
2022-01-06 06:46:02.774027 01/06/2022,06:46:26.26,5.22,-894.53,-8.79,-421.39,24.72,1018.99,21.48,158,
2022-01-06 06:52:02.157746 01/06/2022,06:52:26.26,5.22,-900.39,-2.93,-422.36,24.63,1019.06,21.48,159,
2022-01-06 06:58:02.540408 01/06/2022,06:58:26.26,5.22,-910.16,-9.77,-421.88,24.72,1019.01,21.49,160,
2022-01-06 07:04:01.921364 01/06/2022,07:04:26.26,5.22,-905.27,-6.84,-431.64,24.82,1019.09,21.50,161,
2022-01-06 07:10:02.298420 01/06/2022,07:10:26.26,5.22,-912.11,-5.37,-420.90,24.72,1019.07,21.50,162,
2022-01-06 07:16:02.678596 01/06/2022,07:16:26.26,5.22,-908.20,0.0,-431.64,24.68,1018.96,21.49,163,
2022-01-06 07:22:02.050786 01/06/2022,07:22:26.26,5.22,-895.02,-12.21,-425.78,24.63,1019.08,21.49,164,
2022-01-06 07:28:01.473352 01/06/2022,07:28:26.26,5.22,-908.20,-11.72,-434.57,24.82,1019.08,21.50,165,
2022-01-06 07:34:01.852509 01/06/2022,07:34:26.26,5.22,-896.97,-4.39,-423.34,24.63,1019.01,21.50,166,
2022-01-06 07:40:02.239122 01/06/2022,07:40:26.26,5.22,-912.11,-17.58,-422.36,24.72,1019.03,21.50,167,
2022-01-06 07:46:01.621976 01/06/2022,07:46:26.26,5.23,-905.27,-5.86,-429.20,24.72,1019.0,21.50,168,
2022-01-06 07:52:02.009958 01/06/2022,07:52:26.26,5.23,-898.93,-3.91,-431.15,24.72,1019.12,21.51,169,
**2022-01-06 07:52:36.053366 01/06/2022,07:53:00.38,5.22,-908.20,-8.30,-425.78,24.63,1019.04,21.51,170,<-(34)**
2022-01-06 07:58:36.438268 01/06/2022,07:59:00.38,5.23,-905.27,-18.55,-412.60,24.63,1019.09,21.41,171,
2022-01-06 08:04:35.820607 01/06/2022,08:05:00.38,5.22,-904.30,-10.74,-431.15,24.53,1019.17,21.33,172,
2022-01-06 08:10:36.206776 01/06/2022,08:11:00.38,5.22,-893.55,-9.28,-431.15,24.63,1019.23,21.30,173,
2022-01-06 08:16:35.592003 01/06/2022,08:17:00.38,5.22,-902.83,-6.35,-423.34,24.72,1019.50,21.30,174,
2022-01-06 08:22:35.971991 01/06/2022,08:23:00.38,5.23,-907.23,6.35,-415.53,24.53,1019.55,21.28,175,
2022-01-06 08:28:36.355417 01/06/2022,08:29:00.38,5.22,-897.46,-6.35,-418.46,24.72,1019.48,21.26,176,
2022-01-06 08:34:35.724833 01/06/2022,08:35:00.38,5.23,-911.62,-7.81,-429.20,24.34,1019.59,21.23,177,
2022-01-06 08:40:36.098422 01/06/2022,08:41:00.38,5.22,-897.95,-2.44,-426.27,24.72,1019.56,21.22,178,
2022-01-06 08:46:36.479308 01/06/2022,08:47:00.38,5.22,-901.86,-9.77,-425.78,24.48,1019.45,21.25,179,
**2022-01-06 08:48:31.608826 01/06/2022,08:48:56.37,5.23,-889.16,-14.65,-419.92,24.72,1019.44,21.26,180,<-(116)**
2022-01-06 08:54:31.995066 01/06/2022,08:54:56.37,5.23,-911.62,-17.58,-414.06,24.72,1019.52,21.28,181,
2022-01-06 09:00:32.368889 01/06/2022,09:00:56.36,5.22,-899.41,-1.95,-426.76,24.53,1019.53,21.30,182,
2022-01-06 09:06:31.750427 01/06/2022,09:06:56.37,5.22,-902.83,-6.84,-426.76,24.63,1019.50,21.30,183,
2022-01-06 09:12:32.132255 01/06/2022,09:12:56.37,5.22,-904.79,0.49,-409.18,24.53,1019.61,21.30,184,
2022-01-06 09:18:31.507550 01/06/2022,09:18:56.37,5.22,-912.11,-6.84,-420.90,24.68,1019.66,21.30,185,
2022-01-06 09:24:31.891972 01/06/2022,09:24:56.36,5.22,-909.18,-6.35,-424.32,24.48,1019.69,21.29,186,
2022-01-06 09:30:32.278901 01/06/2022,09:30:56.37,5.23,-909.18,-12.70,-432.13,24.63,1019.67,21.28,187,
2022-01-06 09:36:31.666054 01/06/2022,09:36:56.37,5.22,-914.06,-20.02,-415.53,24.34,1019.79,21.28,188,
2022-01-06 09:42:32.044358 01/06/2022,09:42:56.37,5.22,-898.44,-8.30,-427.25,24.72,1019.79,21.27,189,
2022-01-06 09:48:31.429876 01/06/2022,09:48:56.36,5.23,-892.58,2.44,-425.78,24.39,1019.87,21.26,190,
2022-01-06 09:54:31.811853 01/06/2022,09:54:56.37,5.22,-901.37,-20.02,-421.88,24.29,1019.96,21.27,191,
2022-01-06 10:00:32.201316 01/06/2022,10:00:56.37,5.23,-908.69,-4.39,-425.78,24.72,1019.93,21.34,192,
2022-01-06 10:06:31.588394 01/06/2022,10:06:56.37,5.22,-902.34,-15.63,-432.13,24.91,1019.96,21.41,193,
2022-01-06 10:12:31.978334 01/06/2022,10:12:56.36,5.22,-910.16,-19.04,-420.41,24.63,1019.87,21.45,194,
2022-01-06 10:18:31.390724 01/06/2022,10:18:56.36,5.22,-905.76,-5.86,-411.13,24.82,1019.94,21.49,195,
2022-01-06 10:24:31.777668 01/06/2022,10:24:56.37,5.23,-898.44,-18.55,-417.48,24.68,1019.92,21.52,196,
2022-01-06 10:30:32.158261 01/06/2022,10:30:56.36,5.22,-901.37,-14.65,-420.90,24.53,1019.84,21.54,197,
2022-01-06 10:36:31.538088 01/06/2022,10:36:56.37,5.23,-896.48,-10.74,-426.27,24.72,1019.77,21.57,198,
2022-01-06 10:42:31.918311 01/06/2022,10:42:56.37,5.22,-896.48,-5.86,-426.76,24.87,1019.77,21.59,199,
2022-01-06 10:48:32.292498 01/06/2022,10:48:56.37,5.22,-903.81,-5.86,-422.36,24.72,1019.55,21.61,200,
2022-01-06 10:49:58.390836 01/06/2022,10:50:22.47,5.22,-908.20,1.46,-422.36,25.01,1019.51,21.62,201,
[... 33 lines cut ...]
2022-01-06 13:41:43.628536 01/06/2022,13:42:08.84,5.22,-889.65,-16.11,-425.29,25.01,1017.19,22.01,235,
2022-01-06 13:47:44.011205 01/06/2022,13:48:08.84,5.22,-897.46,-6.35,-416.50,25.25,1016.89,22.08,236,
2022-01-06 13:53:44.402759 01/06/2022,13:54:08.84,5.22,-893.55,-17.09,-418.46,25.92,1016.83,22.24,237,
2022-01-06 13:59:43.791536 01/06/2022,14:00:08.84,5.22,-906.25,-6.84,-430.66,25.92,1016.59,22.39,238,
2022-01-06 14:05:44.180041 01/06/2022,14:06:08.84,5.22,-897.95,-13.18,-430.18,25.92,1016.65,22.52,239,
2022-01-06 14:11:43.600666 01/06/2022,14:12:08.84,5.22,-909.18,-8.79,-425.29,26.30,1016.63,22.66,240,
2022-01-06 14:17:43.984944 01/06/2022,14:18:08.84,5.22,-896.97,-16.60,-432.13,25.73,1016.39,22.77,241,
2022-01-06 14:23:44.369172 01/06/2022,14:24:08.84,5.22,-907.71,-0.49,-446.78,26.21,1016.21,22.84,242,
2022-01-06 14:29:43.751769 01/06/2022,14:30:08.84,5.22,-907.71,-14.65,-420.90,26.21,1015.96,22.92,243,
2022-01-06 14:35:44.124978 01/06/2022,14:36:08.84,5.21,-911.13,-18.07,-424.80,25.92,1015.89,22.98,244,
**2022-01-06 14:36:35.189160 01/06/2022,14:37:00.03,5.22,-287.11,-10.25,-945.31,26.26,1015.92,22.99,245,<-(52)**
2022-01-06 14:42:35.571215 01/06/2022,14:43:00.02,5.22,-815.92,-21.97,-581.05,26.11,1015.99,23.04,246,
2022-01-06 14:48:34.957664 01/06/2022,14:49:00.02,5.22,-821.78,-14.16,-589.84,26.21,1016.03,23.05,247,
**2022-01-06 14:51:48.168120 01/06/2022,14:52:13.35,5.22,-817.38,-6.84,-575.68,26.50,1015.97,23.05,248,<-(193)
2022-01-06 14:56:12.452299 01/06/2022,14:56:37.55,5.22,-803.71,-10.74,-569.82,26.30,1016.03,23.05,249,<-(264)
2022-01-06 14:58:20.597441 01/06/2022,14:58:45.63,5.22,-814.94,-23.44,-583.01,26.30,1015.87,23.05,250,<-(128)**
2022-01-06 15:04:20.982398 01/06/2022,15:04:45.63,5.22,-809.57,-17.09,-576.17,25.92,1015.74,23.06,251,
2022-01-06 15:10:20.369514 01/06/2022,15:10:45.63,5.22,-813.96,-4.39,-587.89,26.26,1015.78,23.06,252,
**2022-01-06 15:13:42.593125 01/06/2022,15:14:07.23,5.22,-830.57,-5.37,-564.45,25.92,1015.82,23.06,253,<-(202)
2022-01-06 15:14:04.626784 01/06/2022,15:14:29.41,5.22,-833.98,-23.93,-568.85,26.50,1015.83,23.07,254,<-(12)**
2022-01-06 15:20:04.002364 01/06/2022,15:20:29.40,5.22,-838.87,-0.98,-562.01,25.73,1015.74,22.90,255,
2022-01-06 15:24:50.296102 01/06/2022,15:25:15.04,5.22,-827.15,-16.60,-547.36,25.44,1015.77,22.72,256,
2022-01-06 15:30:49.678700 01/06/2022,15:31:15.04,5.22,-836.91,-4.88,-545.41,24.87,1015.75,22.50,257,
2022-01-06 15:36:50.064877 01/06/2022,15:37:15.03,5.22,-833.50,-20.02,-550.78,24.82,1015.60,22.36,258,
2022-01-06 15:42:49.444796 01/06/2022,15:43:15.04,5.22,-819.82,-10.74,-542.48,24.72,1015.72,22.20,259,

PaulZC, 2022-01-11:
Hi Ryan,

Please try going back to v1.11. Does it still happen?

A lot changed when we migrated to v2 of Apollo3. Maybe there's a residual gremlin in there somewhere...

Thank you,
Paul

ryanneve, 2022-01-11:
We're also seeing timing problems in 1.11 which is why we tried 2.1. Overall, timing on 2.1 firmware is much more reliable than 1.11. The problem with 2.1 isn't causing us much trouble, just strange.

The log below is from a sensor running 1.11 configured:
- slow sample when pin 11 is high.
- slow sample rate 30 minutes
- fast sample rate 6 minutes
- Minimum awake time 9000 ms
- logging to serial and SD
- has BT on RX/TX, but BT is always on.

As you can see the timing below is very erratic. When this happens it is very hard to get to the menu which I presume means it's ignoring the minimum awake time. Resetting the OLA fixes it for a while.

Here's some data from a sensor in the field running 1.11.
01/11/2022,10:24:27.76,3.86,12.21,-165.04,-1000.98,7.61,1044.96,5.70,1427,
01/11/2022,10:28:52.92,3.99,14.16,-155.76,-1004.88,6.61,1044.91,5.64,1428,
01/11/2022,10:33:51.79,3.99,20.51,-155.27,-994.63,6.46,1045.05,5.80,1429,
01/11/2022,10:38:50.67,3.99,26.86,-146.48,-1005.86,6.46,1044.99,5.79,1430,
01/11/2022,10:43:49.56,3.99,0.00,-156.74,-996.09,6.75,1044.76,5.68,1431,
01/11/2022,10:48:48.45,3.99,10.74,-164.06,-1002.93,6.75,1044.92,5.70,1432,
01/11/2022,10:53:47.33,3.99,28.32,-158.20,-1011.72,6.80,1044.83,5.66,1433,
01/11/2022,10:58:46.21,3.99,6.35,-154.79,-1003.91,6.46,1044.92,5.71,1434,
01/11/2022,11:03:45.10,3.99,16.60,-158.69,-1006.84,6.70,1044.89,5.69,1435,
01/11/2022,11:08:43.98,3.98,12.21,-153.81,-989.75,6.90,1044.65,5.59,1436,
01/11/2022,11:13:42.86,3.99,17.58,-149.41,-998.54,6.75,1044.67,5.61,1437,
01/11/2022,11:18:41.75,3.99,9.28,-153.81,-1008.30,6.61,1044.69,5.62,1438,
01/11/2022,11:23:40.63,3.98,-0.49,-150.88,-1006.84,6.75,1044.81,5.68,1439,
01/11/2022,11:28:39.52,3.98,2.93,-154.79,-1003.91,6.61,1047.28,8.50,1440,
01/11/2022,11:33:38.40,3.98,22.95,-166.02,-1013.18,6.80,1049.08,11.05,1441,
01/11/2022,11:38:37.28,3.98,15.14,-155.76,-1002.93,6.70,1049.84,11.35,1442,
01/11/2022,11:43:36.17,3.98,23.44,-163.57,-1011.72,6.75,1050.43,11.51,1443,
01/11/2022,11:48:35.05,3.98,20.02,-147.46,-1001.95,6.94,1052.73,11.84,1444,
01/11/2022,11:53:33.93,3.98,15.14,-158.20,-1006.35,6.75,1055.51,12.68,1445,
01/11/2022,11:58:32.82,3.98,16.11,-161.13,-1005.86,6.90,1057.08,12.98,1446,
01/11/2022,12:03:31.70,3.98,22.95,-156.74,-998.05,6.94,1058.22,12.99,1447,
01/11/2022,12:08:30.59,3.98,27.34,-154.79,-1010.25,6.90,1059.78,12.95,1448,
01/11/2022,12:13:29.47,3.98,14.65,-166.99,-1006.35,6.94,1060.44,12.88,1449,
01/11/2022,12:18:28.35,3.98,26.37,-160.64,-1002.93,6.75,1061.76,12.85,1450,
01/11/2022,12:23:27.23,3.98,14.65,-148.44,-1009.77,6.94,1063.69,12.80,1451,
01/11/2022,12:28:26.12,3.98,9.77,-156.74,-1006.35,7.23,1065.47,12.77,1452,
01/11/2022,12:33:25.00,3.98,8.79,-152.34,-1001.95,7.14,1066.81,12.74,1453,
01/11/2022,12:38:23.89,3.95,10.25,-160.64,-998.54,6.80,1067.86,12.72,1454,
01/11/2022,12:43:22.77,3.98,17.58,-146.48,-1009.28,6.99,1069.26,12.70,1455,
01/11/2022,12:48:21.65,3.98,14.16,-153.32,-999.02,6.94,1070.55,12.68,1456,

PaulZC, 2022-01-11:
OK - thanks Ryan,

I will try to investigate this - but it may take me a very long time. I have a ton of other projects I need to work through first.

If you have any time to investigate this yourself, and maybe figure out a way to trap these events, I'd really appreciate it.

There is something you can try without changing the code. Please open the debug menu (press 'd' at the main menu) and enable the debug messages. You should see a message saying "goToSleep: sysTicksToSleep (adjusted) = " just before the Artemis goes into deep sleep. Do those values look OK?

Very best wishes,
Paul

PaulZC, 2022-01-20:
Hi Ryan,

Did you have time to try the debug messages? Any clues?

Thanks,
Paul

ryanneve, 2022-01-20:
Paul, Sorry I've been distracted with other things. I'll try to get this tested today or tomorrow.

ryanneve, 2022-01-26:
So this bit of code in lowerPower.ino has a small problem. It only prints out "goTo" before serial output is shut down. Not sure how to best fix. Now way to see how long it's really sleeping for.

```
  if (sysTicksAwake < sysTicksToSleep)
    sysTicksToSleep -= sysTicksAwake;
  else
    sysTicksToSleep = 0;
  printDebug("goToSleep: sysTicksToSleep (adjusted) = " + (String)sysTicksToSleep + "\r\n\r\n");
  
  SerialFlush(); //Finish any prints
  Serial.end();
  Serial1.end();
```

ryanneve, 2022-01-26:
Problem is reproducible, but is on someone else's desk, so I'm debugging at a distance. More soon.

ryanneve, 2022-01-27:
When the OLA goes to sleep, you "disable pins 48 and 49 (UART0) to stop them back-feeding the CH340" in lowerPower.ino.
Is it possible that a device on UART1  (pins 12 & 13), like our Bluetooth module, could cause problems and so should be disabled as well? Something like:
```
if (settings.serialTxRxDuringSleep == false)
{
    am_hal_gpio_pinconfig(48 , g_AM_HAL_GPIO_DISABLE); //TX0
    am_hal_gpio_pinconfig(49 , g_AM_HAL_GPIO_DISABLE); //RX0
    if (settings.useTxRxPinsForTerminal == true)
    {
        am_hal_gpio_pinconfig(12 , g_AM_HAL_GPIO_DISABLE); //TX1
        am_hal_gpio_pinconfig(13 , g_AM_HAL_GPIO_DISABLE); //RX1
    }
}
```

ryanneve, 2022-02-02:
This issue should be fixed by #116

closed 2022-02-02 by ryanneve