sparkfun/OpenLog_Artemis #112: 'class MS5837' has no member named 'begin'

MPH01, 2022-01-20:
### Subject of the issue
Using Arduino 1.8.19 and older versions I'm trying to Verify/Compile unmodified OpenLog_Artemis.ino (and associated files). I've tried a few different SparkFun Apollo3 Boards versions but attached file uses Apollo3 v2.1.0. Board is set to "RedBoard Artemis ATP". See attached file for error and other info, "SparkFun RedBoard Artemis ATP-Apollo v2.1.0 MS5837 no being-error.txt"

MS5837 has (2) version 1.1.0 and 1.1.1. I tried both versions and both do have 'begin'.

Likely this is a version issue of some kind, Arduino IDE, Apollo3, etc. But I can't figure it out. If it compiles okay for you, please state relevant versions used.

### Your workbench
Intended for SparkFun OpenLog Artemis DEV-16832, 'red board'. But isn't required to resolve this issue.

### Steps to reproduce
Open OpenLog_Artemis.ino in Arduino IDE
Sketch>Verify/Compile

### Expected behavior
Sketch>Verify/Compile should not result in any errors.

### Actual behavior
Sketch>Verify/Compile results in error below
[SparkFun RedBoard Artemis ATP-Apollo v2.1.0 MS5837 no being-error.txt](https://github.com/sparkfun/OpenLog_Artemis/files/7902913/SparkFun.RedBoard.Artemis.ATP-Apollo.v2.1.0.MS5837.no.being-error.txt)

exit status 1
'class MS5837' has no member named 'begin'

PaulZC, 2022-01-20:
Hi @MPH01 ,

You have the BlueRobotics MS5837 library installed. The compiler is finding that one first:

```
Alternatives for MS5837.h: [BlueRobotics_MS5837_Library@1.1.1]

ResolveLibrary(MS5837.h)

  -> candidates: [BlueRobotics_MS5837_Library@1.1.1]
```

Delete that library and try again.

Best wishes,
Paul

PaulZC, 2022-01-20:
Hi @MPH01 ,

Best guess - you have another MS5837 library installed and the compiler is finding that one first?

Please look through your libraries and make sure the BlueRobotics one (v1.1.1) is the only one you have installed. Delete any others.

The code is compiling OK for me with v2.2.0 of Apollo3 - RedBoard Artemis ATP.

[image]

Best wishes,
Paul

closed 2022-01-20 by PaulZC

MPH01, 2022-01-20:
Hi Paul,
After manual delete Arduino15 folder, reloading Apollo3 2.2.0 and reload MS5837 from zip from https://github.com/bluerobotics/BlueRobotics_MS5837_Library OpenLog_Artemis.ino (and associated files) compile without errors.

Thanks,
Mica

PaulZC, 2022-01-20:
Hi Mica,
I'm glad that's working for you.
Best wishes,
Paul