sparkfun/OpenLog_Artemis #114: 'Log serial data' nor 'Output serial data' appear to function, Artemis OpenLog v2.1

MPH01, 2022-01-28:
### Subject of the issue
My intent is to pass data from non-OLA controller to OLA via 'Log serial data' feature for recording to mSD. The non-OLA controllers I've tried; Artemis ATP, Artemis Nano and Qwiic Micro SAMD21. If have no indication from OLA any data is received/understood. This is based on exiting terminal option 4, 'Menu: Configure Serial Logging' that states "No user input received. Do you have line endings turned on?" and no modification of behavior/data from terminal window.

Out of desperation I tried the 'Output serial data'. After OLA reboot I can get some garbage text to non-OLA but nothing coherent. As I'm not interested in this functionality I mention it as a possible indication of other problem/error on my part.

As I'm new to serial between controllers I could be doing something wrong; IDE version, IDE controller version installed, what I should see, etc. My ask:

Can someone test 'Log serial data' functionality with OLA v2.1 and verify it works.
  If it works post what I should see, what indicates success.
    I can't find any examples of successful serial data `logging.`
  If it works post non-OLA code and controller used.

I'd post codes I used to send representative non-OLA data but I tried lots of variations and am not sure what OLA is expecting. Which could be the basis of my problem.

### Your workbench
*Arduino 1.8.19
*SparkFun Apollo3 Boards 2.2.0
* Artemis OpenLog v2.1 compiled locally on Windows 7 or Windows 10 from Github download OpenLog_Artemis-main\Firmware
* Tried with and without microSD card.
* Frequency of logging 10Hz, default.
* Tried with and without sensor(s).
* Tried with USB power/connection and battery power to OLA and non-OLA?

### Expected behavior
Unsure.

### Actual behavior
No change in logging behavior based on known serial data being sent.

ryanneve, 2022-02-02:
While we don't use "Log Serial Data", we do use "output serial data" and it has always worked both sending sensor data from the OLA and allowing us access to the OLA menus. Perhaps a baud rate issue? We've got our OLAs connected to a BT module.

PaulZC, 2022-02-02:
Hi Ryan,

I suspect it will be with the pins now being disabled during deep sleep?
Maybe they aren't being reconfigured correctly afterwards?
Whatever is done to TX0/RX0 needs to be done to TX1/RX1...

Cheers,
Paul

PaulZC, 2022-02-02:
Cancel that... I'm mixing up my issues!!

PaulZC, 2022-02-07:
Hi @MPH01 ,

As Ryan says, this does sound like a baud rate issue. Please check that you have the baud rate set correctly and make sure you are using 8 data bits, 1 stop bit, no parity.

Please let us know if you are still having problems.

Best wishes,
Paul

PaulZC, 2022-03-28:
Closing. Please re-open if you still need help with this issue.

closed 2022-03-28 by PaulZC