sparkfun/OpenLog_Artemis #117: OLA waking up from sleep and going into menu when there's a signal on the RX pin.

ryanneve, 2022-02-07:
### Subject of the issue
Despite the firmware change we pushed, we continue to sometimes see OLA waking up when it shouldn't. Note that this isn't always reproducible, but when it gets in this state, it occurs continuously. 

### Your workbench

- OLA with modified 2.1 firmware (#116 ). OLA has pressure sensor on i2c bus.
- BLE UART module is connected to RX/TX and powered by qwiic bus.
- OLA powered by 3x D-cell batteries in series with MEAS cut.
- USB-C not connected.
- usBetweenReadings=40000000 (40 seconds). When we deploy this will be 360 seconds
- qwiicBusPowerUpDelayMs=10000. This is to give BLE module time to power up and connect.
- minimumAwakeTimeMillis=15000. This is to allow data to finish transmitting, and give a window for external access to console.

We have noted that after power is removed from qwiic bus, signal on BREAKOUT_PIN_RX drops slowly from 3v3 to 0. Can provide oscilloscope image if it helps, but I thought the new  `am_hal_gpio_pinconfig(13 , g_AM_HAL_GPIO_DISABLE); //RX1`  before sleeping would make the OLA ignore that pin.

### Steps to reproduce
If we disconnect wire from OLA RX to BLE TX, problem doesn't occur.

### Expected behavior
OLA should stay asleep until woken by timer.

### Actual behavior
Sample is taken.
OLA goes to sleep (as verified in debug statements).
OLA wakes up and goes into menu.

Is there a way to tell what interrupt woke the OLA?

[OLA_settings.txt](https://github.com/sparkfun/OpenLog_Artemis/files/8017979/OLA_settings.txt)
[OLA_deviceSettings.txt](https://github.com/sparkfun/OpenLog_Artemis/files/8017980/OLA_deviceSettings.txt)

PaulZC, 2022-02-07:
Hi Ryan,

I’m wondering if the serial Rx interrupt is still enabled, even though the pin is disabled? The code for this will be buried in the Apollo3 core and won’t be easy to find…

I wonder if a logic buffer will solve your issue? A simple non-inverting buffer that goes high impedance when the power goes off should do the trick? Let me know if you need some suggestions.

All the best,
Paul

ryanneve, 2022-02-08:
Paul,

I'm going to try turning power off to the qwiic bus earlier in the goToSleep() sequence with the hope that any signal from the attached BLE module will be done by the time we go into deep sleep.

The configureSerial1TxRx() function in [OpenLog_Artemis.ino](https://github.com/sparkfun/OpenLog_Artemis/blob/main/Firmware/OpenLog_Artemis/OpenLog_Artemis.ino) sets a pull-up on the RX pin. Should this be disabled before Deep Sleep? This might allow the signal on that pin to drop to ground faster.
Not sure of the syntax to do this.
`am_hal_gpio_pincfg_t (g_AM_BSP_GPIO_COM_UART_RX.ePullup, AM_HAL_GPIO_PIN_PULLUP_NONE);`
It looks to already be re-enabled in wakeFromSleep()

As far as digging into the Apollo3 core.  Perhaps Serial1.end() may not be detaching the interrupts.

I looked some at [Arduino_Apollo3](https://github.com/sparkfun/Arduino_Apollo3), but couldn't find much related to Serial or Serial interrupts. There is this issue of yours [https://github.com/sparkfun/Arduino_Apollo3/issues/423](https://github.com/sparkfun/Arduino_Apollo3/issues/423) which may relate.

I also looked in [ARMmbed/mbed-os](https://github.com/ARMmbed/mbed-os) where I found that [serial_api.c](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Ambiq_Micro/TARGET_Apollo3/device/serial_api.c) has a serial_irq_set function which enables and disables serial interrupts using  am_hal_uart_interrupt_enable() and am_hal_uart_interrupt_disable().
It looks like serial_irq_set() is called from the attach() method in [SerialBase.cpp](https://github.com/ARMmbed/mbed-os/blob/master/drivers/source/SerialBase.cpp). I can't find where to look to see if Serial1.end() detaches its interrupts.

PaulZC, 2022-02-10:
Hi Paul (@paulvha ),

Sincere thanks for this as always - it helps a lot!

Ryan (@ryanneve ): I will try and compile an updated OLA binary for you including these changes. I'm buried in other projects at the moment, but I will get to it as soon as I can. For speed, I might just include Paul's changes directly into the OLA code. Longer term, this needs fixing properly as Paul suggests.

Best wishes,
The other Paul

PaulZC, 2022-03-30:
Hi Paul (@paulvha ),

It has taken a very long time, but I am now trying your fix for this issue. (Thank you again for sharing it!)

I am seeing some strange compilation errors:

```
C:\Users\pc235\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.2.0\cores\arduino\mbed-bridge\core-implement\HardwareSerial.cpp: In member function 'virtual void UART::end()':
C:\Users\pc235\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.2.0\cores\arduino\mbed-bridge\core-implement\HardwareSerial.cpp:128:31: error: 'void mbed::SerialBase::_deinit()' is inaccessible within this context
     UnbufferedSerial::_deinit();  // power down - edit pvh
                               ^
In file included from C:\Users\pc235\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.2.0/cores/mbed-os/drivers/UnbufferedSerial.h:26,
                 from C:\Users\pc235\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.2.0/cores/mbed-os/mbed.h:70,
                 from C:\Users\pc235\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.2.0/cores/arduino/mbed-bridge/Arduino.h:14,
                 from C:\Users\pc235\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.2.0/cores/arduino/sdk/ArduinoSDK.h:9,
                 from <command-line>:
C:\Users\pc235\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.2.0/cores/mbed-os/drivers/SerialBase.h:146:10: note: declared here
     void _deinit(); // paulvha
          ^~~~~~~
C:\Users\pc235\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.2.0\cores\arduino\mbed-bridge\core-implement\HardwareSerial.cpp:128:31: error: 'mbed::SerialBase' is not an accessible base of 'UART'
     UnbufferedSerial::_deinit();  // power down - edit pvh
```

I think I must have missed a step...? But I cannot figure it out. Have I done something silly?

I'm using a fresh install of v2.2.0 and have replaced the three files with the ones from your OneDrive.

Sincere thanks,
Paul

adamgarbo, 2022-03-30:
You know you're in good hands when **_both_** of the Pauls are on the case! :stuck_out_tongue_winking_eye:

Very keen to see the outcome of this issue, especially if it addresses the higher quiescent draw that's been observed on v2.x.

paulvha, 2022-03-31:
My bad, I forgot to include a change. I have updated the document on the share.
**In mbed-os/drivers/UnbufferedSerial.h (1 change)**

Around line 158 add

using SerialBase::_deinit;
 
To the list :
    using SerialBase::readable;
    using SerialBase::writeable;
    using SerialBase::format;
    using SerialBase::attach;
    using SerialBase::baud;
    using SerialBase::RxIrq;
    using SerialBase::TxIrq;
    using SerialBase::IrqCnt;

P.s. I tried to apply a solution from the issue 454, but have not been able to achieve that.. Maybe later

PaulZC, 2022-03-31:
Hi Paul (@paulvha ),

Sincere thanks for this. As my American colleagues would say: "You're the man!"

The fix is working nicely. Activity on the RX pin no longer wakes the Artemis from sleep.

I've got more issues to fix, SPI stops working after coming out of sleep, but I'll investigate and fix that. I suspect I'm not re-enabling something correctly.

Thanks again,
Paul

paulvha, 2022-03-31:
Good to hear..  
With SPI end there were 2 issues: wrong check on dev (https://github.com/sparkfun/Arduino_Apollo3/issues/442) and next to that it should do  dev = 0 in end(), else it will not get a new SPI instance during begin()

Still try to get an alternative working for UART, although that is becoming more complex than the current fix. :-)
regards
Paulvha

PaulZC, 2022-04-01:
OK. I think I've got a working version of OLA which fixes this issue. BUT, to get it working, I've had to regress to v2.1.0 of the core...

Why? Well, with v2.2.0 of the core, it looks like SdFat behaves differently and can leave the CE pin low. This causes all kinds of badness when trying to communicate with the IMU (which shares the SPI bus).

Just to prove I'm not making this up:

Here is the OLA start-up with v2.1.0 of the core (modified with the Serial _deinit fix):

[image]

At 1000ms, there is a burst of activity on the microSD card which lasts for about 270ms. Then the code pauses for about 280ms. Then you can see three bursts of activity while the code reads the settings from the SD card and opens the log files. Then you can see the code initializing the IMU.

Compare this with v2.2.0 of the core (modified with the Serial _deinit fix):

[image]

Notice how the SD card activity is the same BUT the SD CE pin goes low and stays low after the log files are opened. This of course causes all kinds of badness for the IMU and it does not start up correctly.

If I let the code continue, the SD CE stays low until the first file write happens. After that, it seems to recover and the CE appears to function normally, returning to the high state once the activity has finished.

I don't (yet) have a stripped-down example which isolates this. My theory - at the moment - is that something changed in the way SPI operates in v2.2.0, resulting in the strange behavior on the CE pin. Or maybe it is the way digitalWrites or pinModes happen? The CE pin is controlled by SdFat (I'm using v2.1.2 - the latest version). Further diagnosis probably requires digging into SdFat to see how it controls the CE pin and how that could be different with v2.2.0 of the core?

Anyway... I'm going to finish up testing this v2.1.0 hybrid and will hopefully be able to release it in a day or two.

Sincere thanks for all the help @paulvha !

Very best wishes,
The other Paul

PaulZC, 2022-04-01:
Forgot to mention: both tests included the SPI.end changes too:

```
        void arduino::MbedSPI::end() {
            if (dev) {
                delete dev;
                dev = NULL;
            }
        }      
```

paulvha, 2022-04-01:
Recently I wrote a filemanager (based on SdFat) to read a MicroSD card on different Artemis-based boards (https://github.com/paulvha/apollo3/tree/master/Artemis_filemanager). Trying to understand the SdFat code structure was a challenge, but it is an interesting library.  Most routines of SdSpiCard.cpp have the same standard structure: 
```
Do something
Was it good, spiStop(); and return true
Did it fail, spiStop(); and return false 
```
In spiStop(); it will unselect the CE.

But only a few (and this one most noticeable differ) as it seems to be missing spiStop(); when it fails.

```
bool SharedSpiCard::readSectors(uint32_t sector, uint8_t* dst, size_t ns) { 
  if (!readStart(sector)) {
    goto fail;
  }
  for (size_t i = 0; i < ns; i++, dst += 512) {
    if (!readData(dst, 512)) {
      goto fail;
    }
  }
  return readStop();
 fail:
  return false;
}
```
readStop() will send a CMD12 and follow the same standard structure as mentioned earlier.

So maybe the issue is related to an error during reading a sector and this missing spiStop() in the code.

regards,
Paulvha

PaulZC, 2022-04-01:
Thank you Paul - I will give it a try.

Are your changes based on v2.2.0 or v2.2.1?

Have a nice weekend,
Paul

paulvha, 2022-04-01:
I have used V2.2.1.

regards
Paul

PaulZC, 2022-04-08:
Hi Ryan (@ryanneve ),

We have a solution for you. Please try version 2.2 of the firmware. It will be released in a few minutes.

Just for reference:

v2.2 is based on v2.2.1 of the Apollo3 core. That's the latest release which went live a few days ago.
It includes @paulvha 's ```UART:end()``` [modifications](https://1drv.ms/u/s!AvgMkLxXj95qhtVNhdQgRc5AqJcxeA?e=5Bww8W).
I've tested this on V10 and X04 hardware and it works well.
Previously, any activity on the RX pin would have woken the Artemis from deep sleep. The RX pin slowly being pulled low as your BLE module discharged would have done the same thing. Paul's fix prevents this.

I have tested the firmware thoroughly, but it has **so** many options I might have missed something. Please let me know if you see any weirdness.

Hope all is well in Florida - and that you enjoy the new firmware!

Best wishes,
Paul

closed 2022-04-08 by PaulZC

paulvha, 2022-04-08:
PaulZC : will you pass this fix on to Wenn0101/ Kyle so it becomes part of the next release of the Artemis library ?

PaulZC, 2022-04-08:
Sure - not a problem - Kyle ( @Wenn0101 ) is already aware of the changes - thank you!

ryanneve, 2022-04-08:
Thanks! I'll try to get this out in the field (North Carolina) and see how it goes.

PaulZC, 2022-04-08:
Hi Ryan,

Oops! Sorry! Senile moment. Indeed, I hope all is well in NC.

Very best wishes,
Paul