sparkfun/OpenLog_Artemis #118: Feature request: ISO 8601 time format

adamgarbo, 2022-03-27:
Hi @PaulZC,

What are your thoughts on enabling the OLA to log its date and time as a single ISO 8601 string? I know the Americans are fond of their mm/dd/yyyy format, but I think the adoption of an internationally recognized standard could make sense. For those of us who are ingesting the raw data into Python, it'll also help to save some time not having to manually parse the dates and times. 

Relevant xkcd:
https://xkcd.com/1179/

Cheers,
Adam

PaulZC, 2022-03-28:
Hi @adamgarbo ,

I need to revisit the OLA firmware - today as it happens. I'll take a look at this. We'll need to maintain backward-compatibility, but maybe I can add in an extra option for ISO time.

Cheers,
Paul

PaulZC, 2022-04-08:
Added in v2.2. The firmware now supports both yyyy/mm/dd and ISO 8601 formatting. Use menu 2) "Configure Time Stamp" option 5) "Toggle date style" to change the format. The date format will cycle between mm/dd/yyyy, dd/mm/yyyy, yyyy/mm/dd and yyyy-mm-ddThh:mm:ss+hh:mm . With ISO 8601 selected, you can still log the time too - which is useful if you need 0.01s time resolution.

closed 2022-04-08 by PaulZC