sparkfun/OpenLog_Artemis #123: Feature Request: ADS1015

mishal, 2022-04-27:
Hello,

our project is to make the datalogger for the suface water wave measurement. The datalogger was planned to be based on OLA (https://www.sparkfun.com/products/16832 ) with water level sensor (like https://autosen.com/en/Process-Sensors/Level-sensors/Pressure-level-transmitter-AF906) recording signal at 5-10 Hz frequency. 

We realised that the ADC converter which is on OLA creates a important noise of the signal, so we are about to add an extra ADC converter like Qwiic PT100 - ADS122C04, which is supported by OLA but not in stock (and not available for backorder).

The other option would be Qwiic 12 Bit ADC - 4 Channel (ADS1015) (https://www.sparkfun.com/products/15334 ) but it is not supported by OLA and would need to be added into firmware.

Is there another solution? Is it possible to ask you to add the ADS1015 to the OLA?

Thank you

PaulZC, 2022-04-28:
Hello Michal (@mishal ),

The ADS122C04 is currently out of stock everywhere... :-(

Adding support for the ADS1015 is possible (please see [ADDING_SENSORS.md](https://github.com/sparkfun/OpenLog_Artemis/blob/main/ADDING_SENSORS.md)) but we do not have time to do that right now. I will leave this issue open and we will try to include it in the next release.

Apologies,
Paul

PaulZC, 2022-04-28:
The hydrostatic pressure sensor (AF906) is differential - "Pressure compensation with the ambient pressure by means of a capillary tube integrated in the connection cable". I wonder if you could achieve the same thing with the [SDP31](https://www.sparkfun.com/products/17874)? But it might be too sensitive - depending on the height of your waves?

mishal, 2022-05-04:
Thanks for the reply, but SDP31 is not usable for us.
If you could get support for ADS1015 in the next release, that would be great.

Thanks

Fescron, 2022-09-10:
I would also love support for the ADS1015, it's an input module I assumed would work with the OLA (since it can be used quite universally to log (more accurately?) analog signals) but was kinda surprised it wasn't.

PaulZC, 2023-01-14:
Hi Michal (@mishal ) & Brecht (@Fescron ),

Apologies for the very slow response on this. I have added support for the ADS1015 in v2.5. You can log all four channels and/or all four differential channels. You can configure the gain too. I hope it meets your needs.

Best wishes,
Paul

Fescron, 2023-01-14:
No worries! Oh that's very cool thanks! I'll check it out in the following weeks :)

closed 2023-01-14 by PaulZC