sparkfun/OpenLog_Artemis #126: power the OLA with a battery

tnn77, 2022-06-01:
### power the OLA with a battery

### Your workbench
* using Artemis OpenLog v1.19
* connected GND-GND and 3.3V-3.3V using SAFT LSH 20 D-cell battery (tried both with and without Pololu voltage regulator to feed 3.3 V)

### Expected behavior
Should it be powered via GND-GND and 3.3V-3.3V

### Actual behavior
Did not turn on
Used the Lithium Ion Battery - 2Ah battery and worked fine. 

Could you advise?

tnn77, 2022-06-02:
Dear @PaulZC,

Thank you so much for the quick reply. I was sure I tried the reset button, but anyway, this solved the problem. 
I will ensure I use the voltage regular to connect to the 3.3V pin.
Thank you again.

PaulZC, 2022-06-02:
Hi JR,

"Power loss" is detected by a 3.0V voltage supervisor IC, which is connected to PIN_POWER_LOSS (digital pin D3). The supervisor generates [an interrupt](https://github.com/sparkfun/OpenLog_Artemis/blob/6b43745b1acc90154aac9e604117310d43a004c6/Firmware/OpenLog_Artemis/OpenLog_Artemis.ino#L357-L358) which causes the Artemis to go immediately into deep sleep. The only way to wake it again is with a reset.

You can disable the power loss interrupt, but only by editing the firmware.

We did try several ways of having the Artemis check if power had been reconnected, but none were successful. Even with the Artemis in its lowest-power "run" state, running at the lowest clock frequency, it still drew too much current from the back-up battery and caused a full black-out. This in turn causes the Artemis to lose the RTC setting... There are several closed issues discussing exactly this - if you want to wade through them.

If you keep the 3.3V rail above 3.0V, this won't be an issue for you.

All the best,
Paul

PaulZC, 2022-06-02:
@tnn77 - please close this issue again - unless you have something else to add?

jerabaul29, 2022-06-02:
Perfect, many thanks for the explanations! That means that the workflow should be i) program, ii) remove USB, iii) add the power input on the 3.3V rail through the pololu, iv) press the reset button, and all should be ready to go then, right? :) .

@PaulZC do you think it may be possible to provide a firmware version without the low power check, so that there is no way to "brick" a fully enclosed / sealed instrument on which the reset button is not accessible (thinking about a case such as "there is a big shock or hump, then there is a very short bad contact on the batteries, and instead of just restarting with a bit of information loss which would be ok, the whole OLA gets effectively locked / bricked inside the fully closed box")? May be a good idea to add a pad that can be cut to disable the low sleep power feature and / or the backup battery on future models? :) .

@tnn77 : will need to make sure in the instructions to indicate how to ensure that the OLA logging is started when the batteries are inserted - any LED blinking or something like this that can be used as an indication? :)

PaulZC, 2022-06-02:
Hi JR,

Based on previous discussions and issues, for most users keeping the RTC running is more important. And the only way to ensure that is to put the Artemis into deep sleep.

However, I appreciate your needs are quite different. For now, can I suggest you create a fork of the firmware that meets your needs? Or raise a new "feature request" issue and I will create a custom OLA binary for you - with the power loss interrupt disabled. For safety, you probably need an auto-reset via the Watchdog Timer too...

We are working on a project which will allow you to create your own OLA-like logging system simply and easily on any of a wide range of hardware platforms. And without needing to include multiple libraries. I can't say more about it at present, but it is "in the works". ;-)

Best wishes,
Paul

ryanneve, 2022-06-02:
Paul,
The option to disable the power loss interrupt might be nice. After multiple occasions of mysterious shutdowns when there was plenty of power, I've made it standard practice to disable the powerLossISR() interrupt. This seems to prevent the mysterious un-recoverable shutdowns. I'll open an issue.
-Ryan

ryanneve, 2022-06-02:
As an aside, It would be interesting what functionality could be implemented if the backup battery was bigger. It should be possible to replace the current battery with a bigger one and test this.

PaulZC, 2022-06-02:
Hi Ryan (@ryanneve ),

One 'fix' is to remove the back-up battery altogether. That way the Artemis will automatically reset when the power returns. But, true, a bigger battery, _which can deliver more current_, would allow the Artemis to periodically wake and see if the power has been reconnected.

@tnn77 / @jerabaul29 : I'm closing this now that you've opened #128. I'm neck deep in the new logging library at the moment. I'll try and compile the new binary for you as soon as I have a spare few hours.

All the best,
Paul