sparkfun/OpenLog_Artemis #127: Bug/Feature Request - PIN_POWER_LOSS triggered when it shouldn't be.

ryanneve, 2022-06-02:
### Subject of the issue
Occasional shutdowns when deployed, probably due to PIN_POWER_LOSS interrupt when there should be plenty of power.

### Your workbench
OLA powered by 3xC cell alkaline batteries. Sampling every 6 minutes. External BLE module and pressure transmitter powered by Qwiic bus. Should have enough power for many months.

### Steps to reproduce
Not usually reproducible, but on several occasions after several days of operating fine, the OLA has mysteriously shut down. Recorded battery voltages are well above minimums. Due to significant effort required to physically reach sensor and reset, the benefits of this functionality are outweighed by the liabilities.

### Expected behavior
Should only trigger on actual low battery.

### Actual behavior
Triggers mysteriously. I've been disabling this functionality and re-compiling firmware. It might be nice to have a menu option to disable, or perhaps perform additional voltage checks to confirm low battery situation.

PaulZC, 2022-06-02:
Hi Ryan (@ryanneve ),

Catching and debugging the 'glitch' shut-downs will be extremely challenging. If it only happens after ~days, I would need a dedicated set-up to catch it - almost certainly with an oscilloscope or a logic analyzer running full time and monitoring PIN_POWER_LOSS.

Is there any chance it could be (e.g.) lightning related?

I'd like to close this issue in favor of #128, if that's OK? I've got no problem with compiling the custom binary, but trying to catch the glitch is not something I can do any time soon.

Best wishes,
Paul

ryanneve, 2022-06-02:
@PaulZC , I agree it would be almost impossible to catch. The OLAs with the mysterious shutdowns are in waterproof cases, down in urban storm drains (sometimes several hours drive away). Unlikely to be lightning. What are your thoughts on the OLA doing some sort of sanity check on the battery voltage before calling powerDownOLA() if (powerLossSeen)? Feel free to close this although my goal is always to get away from compiling my own version of the code. As an alternate solution, I could try implementing a menu selectable IgnorePinPowerLoss flag and pushing that change.

PaulZC, 2022-06-02:
Hi Ryan,

When the power loss interrupt is triggered, the voltage is (supposedly) already below 3.0V and falling fast. The Artemis needs to go into deep sleep as quickly as possible, otherwise it starts drawing too much current from the back-up battery, blacks-out and resets the RTC. Spending additional time measuring the voltage and deciding whether to go into deep sleep is - IMO - a risky thing to do.

I should be able to compile the modified binary in a couple of days. But I have some other work I need to finish first...

Closing in favor of #128 - but please re-open or continue the discussion if you think it would be useful.

Best wishes,
Paul

closed 2022-06-02 by PaulZC