sparkfun/OpenLog_Artemis #128: Feature request: firmware pre compiled binaries, without the interrupt driven low power sleep

jerabaul29, 2022-06-02:
Follow up of https://github.com/sparkfun/OpenLog_Artemis/issues/126 : 

Can we get some pre compiled firmware binaries that deactivate the power loss interrupt, so that in case of a transient power interruption for example due to vibrations, the OLA just stops / brows out temporarily, and then recovers / reboots when the power is back, rather than get stuck in a deep power sleep?

Use case: uses of OLA that need to log, "no matter what", but where reboots are not such a big issues, and are inaccessible for user reset button pressing.

FYI @tnn77

adamgarbo, 2022-06-02:
@jerabaul29 

In case you weren't aware, this is something you can do yourself from the Arduino IDE:

`Sketch > Export compiled Binary`

Cheers,
Adam

jerabaul29, 2022-06-02:
@adamgarbo thanks :) . Yes, I know, this is already what I do with the OMB firmware :) . The idea here is that I have not set up the libraries in the right versions to compile the OLA firmware, and I am not too familiar with the OLA firmware either, so would feel 'safer' if someone very familiar with this specific project tales care of the setup and configuration and compilation :) .

jerabaul29, 2022-08-06:
Any update on this? In addition to firmware update / different versions, could an option be to 'simply' add a connection pad between the coin cell battery and the Artemis, so that the coin cell battery can be disabled altogether if wanted?

ryanneve, 2022-08-06:
@jerabaul29 
I've been disabling this behavior in firmware after our OLAs have gone to sleep and never woken up. Resetting the OLA fixes it, but my OLAs are very hard to get to. Batteries seem to always be fine. It's a mystery. See #127 
Unfortunately, I've got a new PC so need to re-setup my development environment. Will try to get a change pushed that adds a configuration option to ignore low voltage sensor interrupt.

PaulZC, 2022-08-07:
This is still on the things-to-do list. I haven't forgotten, I just have a ton of other projects to tackle first...

Sorry for the delay,
Paul

jerabaul29, 2022-08-16:
Amazing, many thanks!

I wont have the hardware to test, but maybe @tnn77 can? :)

PaulZC, 2022-08-17:
Hi Ryan,

But if we do that, the code will almost certainly need a reset to wake again. powerDownOLA puts the Artemis into a deep sleep state. If there is any charge in the back-up battery, the battery will keep it in that deep sleep state... Only a reset will wake it. Sounds ~risky to me?

Thanks!
Paul

ryanneve, 2022-08-17:
@PaulZC , you may be right. My thought was that transient power issues probably won't appear while the OLA is in setup(). Let's try it your way and see how it goes. Is the goal to change this to a configuration option eventually? Excited about the watchdog timer!

PaulZC, 2022-08-17:
Configuration options are nice, but, for critical stuff like this, I always lean towards having it hard-coded. Then there's ~zero possibility of the WDT being mis-configured. The WDT is there to help save you from code-corruption during a brown-out. If running the WDT is selective, there's a small but non-zero chance of it being disabled when you really need it.

Please give the binary a thorough shake, whenever you can spare the time. I'm grateful for the feedback.

Very best wishes,
Paul

tnn77, 2022-08-17:
@jerabaul29 
Yes, I have one at my desk. I'll try to find time to test the binary. I'll let you know when I get a chance!

PaulZC, 2022-08-26:
This feature has been included in v2.4. Going forward, we will include ```noPowerLossProtection``` binaries for each new release.

closed 2022-08-26 by PaulZC