sparkfun/OpenLog_Artemis #129: Feature request: support the ISM330DHCX IMU over qwiic

jerabaul29, 2022-06-15:
### Subject of the issue

Would it be possible to support logging hte ISM330DHCX IMU over qwiic? This IMU is available with a qwiic port from different sources, for example:

- https://www.adafruit.com/product/4569
- https://www.st.com/en/evaluation-tools/steval-mki207v1.html (quite a few resellers)
- https://www.adafruit.com/product/4502

Given the limited supply of the IMU that is present by default on the OLA, adding an external IMU over qwiic would be great. In addition, the ISM330DHCX is in my experience one of the highest quality, fully temperature compensated, IMU / 6dof / 9dof sensor I know of.

jerabaul29, 2022-06-15:
(PS: seems like there are some good "driver" libraries for it; see for example: https://github.com/stm32duino/ISM330DHCX , or the Adafruit library adapted by https://github.com/jerabaul29/OpenMetBuoy-v2021a/blob/main/development_environment/setup_arduino_v1-8_environment/Instructions.md ).

jerabaul29, 2022-07-16:
I see that the ISM330DHCX is now available as a SF breakout, amazing :) . https://www.sparkfun.com/products/19764 Support out of the box for the OLA would be super useful (cc @tnn77 ).

jerabaul29, 2022-08-06:
@PaulZC , do you have some plan to support the ISM330DHCX (and the magnetometer included on the 9dof breakout using it :) ) in the OLA firmware? :)

PaulZC, 2022-08-07:
Hi JR,

I think I've dropped hints about this before. We're working on a new logging product which will have an ISM330 and a Mag on-board. It is designed to work with the new Qwiic Universal Auto Detect library. Coming soon-ish...

Best wishes,
Paul

PaulZC, 2022-08-17:
After discussing this with my "higher authority" ;-) , we'll leave this on the things-to-do list. But lower down the list than OpenLog ESP32 - which has the ISM330 and Mag on-board, interfaced via SPI for very fast logging speeds.

jvoermans, 2022-08-22:
Hi Paul,

I'm also interested in support of the ISM330DHCX with the OpenLog. It will make the OpenLog my go-to datalogger in my laboratory experiments (if I don't need remote data transmission).

Also, any plans to have BMP388 and/or BMP390 support?

Kind regards,
Joey

jerabaul29, 2022-08-26:
@PaulZC just curious: what is the OpenLog ESP32? A new OpenLog based on another chip? Any more reason for moving from Ambiq to ESP32? I must say I like these small ambiq quite well, specially due to their incredibly low power consumption, so if possible to support ISM330DHCX with OpenLogArtemis, this would be great :) .

jerabaul29, 2022-08-26:
(PS: for us really high power effectiveness means quite a lot, and since the ISM330DHCX is very power effective, and the Ambiq is very power effective, this is a really nice combo for us :) )

PaulZC, 2022-08-26:
Hi JR,
We do still plan to add support for the IMU330 and MMC5983 (Mag) to OLA, but the new OpenLog ESP32 is being given priority first. (I've only got 10 fingers...!)
**Spoiler alert!**
OpenLog ESP32 will be based on the ESP32 WROOM 32E. It's sleep current won't be as low as OLA. But that's outweighed by: built-in WiFi and BLE support; and being able to connect the microSD via full 4-bit SDIO. The IMU330 and MMC5983 will be interfaced via SPI, meaning you'll be able to log both at their maximum rates (6.7kHz and 1kHz) should you want to. Qwiic Universal Auto-Detect will make it _much_ easier to add new sensors like the BMP3nn parts. I'm also looking forward to being able to log GNSS data at high rates via SPI and SDIO too.
Coming soon! (Ish...)
Best wishes,
Paul

jerabaul29, 2022-08-26:
Many thanks for the update and your hard work on this :) . Perfectly understand that you have 10 fingers and a limited number of working hours per week :) .

Just to give some context. For some of the applications (like lab experiment), power consumption is not an issue, and the ESP32 sounds really good :) . But for some other applications (like putting the logger on a box on an ice floe in the middle of the Arctic for a few weeks, and picking it up later on), we really enjoy the low power consumption of the Artemis, and we are fine with logging IMU at only 100 or even 50 Hz :) . Not a criticism, just explaining why there are cases where we like the OLA :) .

PaulZC, 2023-01-14:
Hi JR (@jerabaul29 ) & Joey (@jvoermans ),

Apologies for the very slow response on this. I have added support for the ISM330DHCX in v2.5. You can configure the scale and rate for both the accel and gyros, and configure the filters too. I hope it meets your needs. I've also added the MMC5983MA magnetometer, just in case you need Mag data too.

Best wishes,
Paul

closed 2023-01-14 by PaulZC

jerabaul29, 2023-01-14:
Many thanks! Also FYI @tnn77 , this way can use the OLA with the ISM330DHCX out of the box :) .

tnn77, 2023-01-14:
@jerabaul29 this is great news! thank you so much @PaulZC