sparkfun/OpenLog_Artemis #133: OpenLog Artemis Without IMU - not detecting qwiic devices

rglongoria, 2022-08-10:
### Subject of the issue
I have just received one of these new models without the IMU. Can get it to log time stamp and sample rate but it will not detect any qwiic sensors I attach, even though I'm using one listed on the list Sparkfun provided.

### Your workbench
* Are you using a microSD card? 1 GB
* At what frequency are you logging? For example, 10Hz (default)
* What version of firmware are you using? Artemis OpenLog v1.10
* How is OpenLog Artemis wired to your sensor(s)? qwiic
* How is everything being powered? via USB-C and have also tried a LiPo battery
* Are there any additional details that may help us help you?

### Steps to reproduce
All I've done is log into the serial monitor to view logged data. Have also looked at data files on the SD card. Nothing is being logged except time stamp and the data rate.

In Serial Monitor, I can run the menu and have run '6' to detect devices. Comes back that no devices are connected. I have a Qwiic Pressure MS5637 connected.

### Expected behavior
Expect to see pressure data

### Actual behavior
No pressure data is being logged, device is not detected

PaulZC, 2022-08-10:
Hi @rglongoria ,

I'm sorry you are having problems with the OLA.

I have just tested an IMU-less OLA here with an MS8607 sensor and it is working fine. The MS8607 is very similar to the MS5637, except it has an extra humidity sensor built-in (with its own I2C address).

You say you are running Artemis OpenLog v1.10 firmware? Are you sure? All of the IMU-less OLAs ship with version 2.2 firmware installed. v1.10 is really very old. Can you please check the version again?

Thank you,
Paul

rglongoria, 2022-08-10:
Hello Paul,
sorry, it is running Artemis OpenLog v2.3
By the way, this device does not start up right away. No status lights, etc.
I need to reset a few times before it starts logging.

PaulZC, 2022-08-10:
If the OLA has gone into deep sleep, it needs a reset to wake it up. There is more information about this in the hook-up guide:

https://learn.sparkfun.com/tutorials/openlog-artemis-hookup-guide/low-power-considerations

This is very puzzling!

I have also tested an MS5837 - again very similar to the MS5637 - and it too works OK. Unfortunately I don't have an MS5637...

Do you have a different sensor you can test?

If you are willing to do a little diagnostic work: please open the menu and select the secret option "d" to open the debug menu. Please enable the debug messages, exit the menus, then restart the OLA. Please copy and paste the start-up messages here. They may provide some clues. It should look like the text below:

Or we can arrange a return / replacement for you. Just let me know.

```
Artemis OpenLog v2.3
Logging to: dataLog00061.TXT
Logging to: serialLog00002.TXT
ICM_20948::checkID: ICM_20948_check_id returned: Wrong ID
ICM_20948::startupDefault: checkID returned: Wrong ID
ICM_20948_SPI::begin: startupDefault returned: Wrong ID
beginIMU: first attempt at myICM.begin failed. myICM.status = 4
ICM_20948::checkID: ICM_20948_check_id returned: Wrong ID
ICM_20948::startupDefault: checkID returned: Wrong ID
ICM_20948_SPI::begin: startupDefault returned: Wrong ID
beginIMU: second attempt at myICM.begin failed. myICM.status = 4
ICM-20948 failed to init.
SD card online
Data logging online
Serial logging online
IMU offline - or not present
detectQwiicDevices started
waitForQwiicBusPowerDelay: qwiicPowerHasBeenOnFor 1040ms
Identifying Qwiic Muxes...
detectQwiicDevices: something detected at address 0x76
detectQwiicDevices: MS8607/MS5637/MS5837/BME280 found at address 0x76. Ignoring it for now...
Identifying Qwiic Devices...
detectQwiicDevices: something detected at address 0x76
detectQwiicDevices: added Pressure-MS5837 at address 0x76
Autodetect complete
waitForQwiicBusPowerDelay: qwiicPowerHasBeenOnFor 1130ms
beginQwiicDevices: attempting to begin deviceType Pressure-MS5837 at address 0x76 using mux address 0x0 and port number 0
beginQwiicDevices: device is online
setMaxI2CSpeed: setting qwiic clock speed to 100000Hz
Pressure-MS5837 online at address 0x76
Device count: 1
rtcDate,rtcTime,mbar,degC,depth_m,alt_m,output_Hz,
01/01/2000,00:29:17.19,1020.07,28.80,0.072,-56.59,20.0,
01/01/2000,00:29:17.29,1020.10,28.85,0.073,-56.84,13.333,
01/01/2000,00:29:17.40,1020.05,28.88,0.072,-56.43,12.0,
01/01/2000,00:29:17.50,1020.04,28.89,0.072,-56.35,11.429,
01/01/2000,00:29:17.60,1020.05,28.90,0.072,-56.43,11.111,
01/01/2000,00:29:17.70,1020.06,28.91,0.072,-56.51,10.909,
01/01/2000,00:29:17.80,1020.03,28.91,0.072,-56.26,10.769,
```

rglongoria, 2022-08-10:
I've tested with several qwiic devices I have, same results. 

Here you go after debug enabled:

0a00,07:Artemis OpenLog v2.3
Logging to: dataLog00020.TXT
Finding the next available log file.
This could take a long time if the SD card contains many existing log files.
Logging to: serialLog00000.TXT
ICM_20948::checkID: ICM_20948_check_id returned: Wrong ID
ICM_20948::startupDefault: checkID returned: Wrong ID
ICM_20948_SPI::begin: startupDefault returned: Wrong ID
beginIMU: first attempt at myICM.begin failed. myICM.status = 4
ICM_20948::checkID: ICM_20948_check_id returned: Wrong ID
ICM_20948::startupDefault: checkID returned: Wrong ID
ICM_20948_SPI::begin: startupDefault returned: Wrong ID
beginIMU: second attempt at myICM.begin failed. myICM.status = 4
ICM-20948 failed to init.
SD card online
Data logging online
Serial logging online
IMU offline - or not present
detectQwiicDevices started
waitForQwiicBusPowerDelay: qwiicPowerHasBeenOnFor 550ms
waitForQwiicBusPowerDelay: delaying for 450
Identifying Qwiic Muxes...
Identifying Qwiic Devices...
No Qwiic devices detected
rtcDate,rtcTime,output_Hz,
01/01/2000,07:51:33.40,0.0,
01/01/2000,07:51:33.50,20.0,
01/01/2000,07:51:33.60,15.0,

PaulZC, 2022-08-10:
It could be the Qwiic cable. Have you tried different cables?

Apart from that, sorry, I'm completely out of ideas... All I can suggest is that you return your OLA and we will arrange a refund or replacement.

If you bought it from a distributor, please arrange a return/refund through them. If you bought it direct from SparkFun, please open a return ticket here: https://www.sparkfun.com/returns . Please cut and paste this link into the "Why do you want an RMA?" box: https://github.com/sparkfun/OpenLog_Artemis/issues/133

Many apologies for the inconvenience,
Paul

rglongoria, 2022-08-10:
Yes, will return. thanks

----------------------------------------------------
Raul G. Longoria, Ph.D., P.E. | Professor | Walker Department of Mechanical Engineering
The University of Texas at Austin | Austin, Texas 78712
http://sites.utexas.edu/longoria/

PaulZC, 2022-08-10:
Closing... Please re-open if you need any more help with this.

closed 2022-08-10 by PaulZC