sparkfun/OpenLog_Artemis #134: Feature request: detach logging rate from peripheral update rate

caqt-cc1, 2022-08-15:
### Subject of the issue

When using a slow to update measurement unit, such as a GPS module, the logging rate becomes limited by this slow update rate whilst other devices are able to update much faster.
This is further compounded when such a device halts the datastream, I assume to update its value.
I would like to propose that the datarate remain the same, and the slower to update devices return their last updated value until a new value is given.

### Your workbench
Arduino Serial Monitor, 115200 baud.
PC - USB C - OpenLog Artemis - qwiic cable - ZOE-M8Q module
OLA Firmware version 2.3

### Steps to reproduce
Connect a GPS module such as the ZOE-M8Q and compare the terminal update rate when the GPS module is not connected.

### Expected behavior
Terminal Log rate remains at rate set to. Values from measurement units remain static until updated.

### Actual behavior
Data rate is limited to slowest updating device.

PaulZC, 2022-08-15:
Hi @caqt-cc1 ,

Using the GNSS "Automatic" option will solve this.
Open the menu, and open option 6 (Qwiic devices).
Select the GNSS from the list.
Enable the "Use autoPVT" option.
Exit the menus.

Best wishes,
Paul

closed 2022-08-15 by PaulZC

caqt-cc1, 2022-08-15:
After some further testing, I have found that this setting resets back to default when the OLA is reset (with either the reset button or opening a new terminal connection). Is there a way to save this setting?

PaulZC, 2022-08-15:
It _should_ be being saved automatically in the software settings - either to microSD card or EEPROM or both. 

https://github.com/sparkfun/OpenLog_Artemis/blob/6801bbe5db68d276c38c19bd9a90896d907bf80f/Firmware/OpenLog_Artemis/settings.h#L136

But maybe I've missed something... I will try and find time to investigate this later today.

You do need to make sure you "Exit" the menus correctly, otherwise the updated settings are not saved. Please make sure you always keep pressing "x" until the code has gone back to logging data after you make the change.

Best wishes,
Paul

PaulZC, 2022-08-15:
I've tested this and - for me - it is saving the setting correctly. autoPVT is still "Yes" after resetting the OLA. But I am using a microSD card...

Please make sure you "Exit" the menus correctly, otherwise the updated settings are not saved. Please make sure you always keep pressing "x" until the code has gone back to logging data after you make the change.

If you are not using a microSD card, then, unfortunately, the device settings _are_ lost after a reset. The system settings (logging interval) etc. are saved in EEPROM and to microSD card. But the device settings are _only_ saved to microSD card. autoPVT is a device setting...

Can you please confirm if you are using a microSD card?