sparkfun/OpenLog_Artemis #139: Feature Request: Support AMG8833 Grid-EYE Infrared Array Breakout

jackmcnamarauq, 2022-10-14:
Would it be possible to add support for the SparkFun Grid-EYE AMG8833 board? There are pre-existing libraries and I would like to be able to log the teamperature data provided with an OpenLog Artemis. The board already supports the qwiic connectors also.

Thanks

PaulZC, 2022-10-14:
Hi @jackmcnamarauq ,

It's a very interesting idea. I'm not sure it's a good match for OpenLog Artemis, due to the amount of data that would need to be logged. But it will be straightforward to add to our new device driver framework. We will be launching a new OpenLog product in the coming weeks, which will ship with the framework pre-installed. I will make sure it is on the list for that.

Best wishes,
Paul

PaulZC, 2022-10-19:
Hi @jackmcnamarauq ,

The [SparkFun GridEYE AMG8833 library](https://github.com/sparkfun/SparkFun_GridEYE_Arduino_Library) has just had a major update. I've resolved all of the issues with I2C communication and negative temperatures. If you have time, please take a look.

I've also today added the AMG8833 into our new "framework". The framework isn't public yet, but will be soon. At the moment, I'm returning the 64 pixel values as floats (1 decimal place) in Comma Separated Value format. This is the same format that OpenLog Artemis uses. But I'm wondering if a different format would be better for this kind of data?

May I ask, how do you plan to use the pixel data from the GridEYE? Is floating point CSV OK for you? We can offer you JSON too. Or would some kind of simple "image" format be better?

Thanks,
Paul

jackmcnamarauq, 2022-10-20:
Hi Paul,

Apologies for the delayed response. That’s great to hear there’s been an update to the AMG8833 library and I’ll have a look at that. I think your suggestion of floats (1 decimal place) is perfect. We plan on using the AMG8833 alongside a Human Presence Detection sensor to verify that board and log temperatures around a desk using the OpenLog Artemis. I don’t think JSON or Image formats would be necessary.

Cheers,
Jack

closed 2024-02-21 by PaulZC