sparkfun/OpenLog_Artemis #143: Scheduled new file

Mikxie, 2022-12-19:
Would it be possible to make a new logfile at a given time in stead of at a given interval. It would be nice to have one log for each day, or a new log every day at noon.

PaulZC, 2023-01-03:
Hi @Mikxie ,

This is possible - but it is a non-trivial code change. The complication comes when sleeping between measurements. We would need to adjust the code to ensure the board wakes at the same time each day to open the new file.

I will add this as a feature request, but I can't guarantee if/when we will be able to implement it. We are working on another, more sophisticated logger product and that is taking up resources at the moment.

Best wishes,
Paul

PaulZC, 2024-02-21:
Closing... Sorry! It's just not practicable to add this feature to the existing firmware. It would require substantial code changes. Please check out the [DataLogger IoT](https://www.sparkfun.com/products/20594). I'm sure that supports log file changes.

Best wishes,
Paul

closed 2024-02-21 by PaulZC