sparkfun/OpenLog_Artemis #144: Feature request: Support for high-G acceleromters - KX134, H3LIS331DL, or equivalent

VancouverUmbrella, 2022-12-29:
Hey there,

Would you consider adding support for a high-G accelerometer such as the following?

- [SparkFun Triple Axis Accelerometer Breakout - KX134 (Qwiic)](https://www.sparkfun.com/products/17589)
- [SparkFun Triple Axis Accelerometer Breakout - H3LIS331DL](https://www.sparkfun.com/products/14480)

**My application**
I'd like to log acceleration data at high frequency for radio-controlled discus launch gliders (DLG). Such model aircraft are launched with an acceleration calculated at 50G or higher resulting in speeds believed to exceed 100mph in some cases. However, obtaining data on speed has proven elusive and error-prone. The ultimate goal is to log data that can be used to better understand the factors that affect launch speed -- and ultimately launch height. The higher the frequency of the data, the better. 10Hz would likely be perfectly usable. The OLA sure seems like it could provide an easy and very effective 'out of the box' solution.

**Past experience**
Last year, I successfully used an OLA to log altitude data of model aircraft for a different project. The ease of use of the OLA -- and the quality of the logged data -- made it a great choice.

Many thanks,
Andrew

PaulZC, 2023-01-03:
Hi Andrew (@VancouverUmbrella ),

This is a good request - especially for the IMU-less version of the OLA.

However, we are working on a new, more sophisticated logger product and that is soaking up resources at present. I will leave this open but I can't guarantee if/when we'll be able to implement it.

Best wishes,
Paul

PaulZC, 2023-01-14:
Hi Andrew (@VancouverUmbrella ),

I've added support for the KX134 in v2.5. Please give it a try - it should meet your needs.

It defaults to 50Hz 64g mode. You can change the range to 8/16/32g if needed - via the device menu. You can also put it into 400Hz mode, just in case you need the extra sample speed.

The way the code works is this:
* The code checks if the KX134 has new data available and updates the ```dataReady``` flag
* If new data is available, it is read via I2C
* The data logged to file is: fresh (new) if dataReady is 1, stale (previous / duplicated) if dataReady is 0

This allows you to log data as fast as possible, without stalling, even though some of the data logged could be stale / duplicated.

If you set the logging interval to the standard 0.1 seconds (10Hz), you'll get fresh / new data every sample. Even with the sensor running at 50Hz.

If you set the logging interval to 0.0025 seconds (400Hz), you'll get fresh / new data typically every other sample due to the overheads of reading the data via the I2C bus, writing it to SD card and printing it to the terminal. The dataReady flag allows you to discard stale (prevous / duplicate) data if you want to. This way you maximise the sample rate and stand the best chance of capturing your high-g events.

Disable the "log to terminal" option for the highest sample rates.

If you need a more detailed description of how this works, let me know.

Have fun!

Best wishes,
Paul

PaulZC, 2023-01-14:
PS. Please post some of your logged data here, when you have the chance. We've got RC aircraft enthusiasts at SparkFun who'd love to see that data... Or, please post a link if you normally share that data elsewhere. Thanks!

VancouverUmbrella, 2023-01-14:
That's outstanding, Paul! Thank you very much!

I'll give the KX134 a try this weekend. If the rain stops here, I might even be able to post some real data. 

By the way, the data I've obtained with the OLA so far has proven very interesting. As expected, acceleration was off the scale of the onboard accelerometer. However, the gyro data showed an unexpectedly extreme yaw event in the factions of a second after release. It seems the model was practically flying sideways for a brief instant. Plus, data from a BMP280 showed a pressure drop equivalent to about -20m of altitude in the most extreme cases.

I'll be back with an update asap. And thanks again for your wonderful help.

Best,
Andrew

VancouverUmbrella, 2023-01-16:
Hey again,

The skies cleared yesterday and I got out for an hour of launch tests with the new OLA firmware. There were no problems to report. A ton of very interesting data was gathered. The plots below give a taste:

**BELOW** - An overview of all the launches in the session.
[image]

**BELOW** - Acceleration in the longitudinal axis of the glider (aY) as logged from the KX134. The KX134 was configured for +/-64g. Altitude was derived from a BMP280 connected to the OLA.
[image]

**BELOW** - Acceleration in the latitudinal axis of the glider (aX). Ouch! It seems 64g isn't enough to capture the full range. Not even close. The acceleration data spans just under +/-8 units, from which I infer that each unit is 8g. Did I get that right?
[image]

**BELOW** - Acceleration in the vertical axis of the glider (aZ). Out of range again. :(
[image]

**BELOW** - Angular acceleration around the vertical axis (gZ) as reported by the IMU gyro. Wow! The peak is more than 1900 degrees/s.
[image]

Some tentative conclusions:

- A DLG launch involves some pretty darn big accelerations and angular velocities! I'm surprised just how big. +/-64g doesn't cover the range in some axes. I'll take a look at your firmware code, Paul, to see if I might be able to add OLA support for a higher G accelerometer such as the H3LIS331DL.
- I was wrong in thinking 10Hz would be sufficient. 100Hz is more like it.
- The log data will likely yield more useful info. I'll buckle down to see if it's possible to derive the yaw angle immediately after release, and perhaps an approximation of velocity. Perhaps that latter dream is optimistic given the challenges of integrating accelerometer data over time.

I've attached an Excel file with the log data for launch no. 8. The log for the full session is too big to attach here. The zip also includes my OLA configuration files from the test session.

Many thanks again for the great support for the OLA. This stuff is fun!

Best,
Andrew
[Hawk-kx134-20230114-launch8.zip](https://github.com/sparkfun/OpenLog_Artemis/files/10422699/Hawk-kx134-20230114-launch8.zip)

PaulZC, 2023-01-16:
Hi Andrew (@VancouverUmbrella ),

This is excellent stuff - thank you.

Looking at your data, I'd say the KX134 is possibly in 8g mode, not 64g mode... I've taken at quick look at our library source code and it looks like we might have some gremlins in there. I'll dig into this a bit more and will report back here.

Thanks again for sharing. I love this stuff.
Paul

VancouverUmbrella, 2023-01-16:
Hey Paul, that would actually be a wonderful bug from my point of view. It holds out hope that 64G might cover my project after all.

I'm having a hard time swallowing the 1900 dps rotation, too. That seems awfully high to me. Any chance you could keep your eye peeled for any issue there, too? Of course, the 'problem' may actually be in my expectations!

All the best,
Andrew

VancouverUmbrella, 2023-01-16:
Great finds, Paul.

In working on my own implementation of a KX134 + Teensy 4.1, I came across some other issues in the KX13xx library. Now that you have the KX134 running with the OLA, I happily set my more limited approach aside.

Regarding the gyro data, I'm starting to think that it may actually be OK. That cautious conclusion is based on integrating the velocity over time to arrive at the total rotation. In the first plot below, that total rotation from the launch phase to flight works out to a smidge under 500 degrees. That's exactly the sort of number I would have expected. 

**BELOW** - Integration of angular velocity (gZ) to infer total rotation.
[image]

Then looking at total rotation, the first yaw event after release seems to be about 40 degrees. I think that's a higher number than the DLG community generally accepts. Then again, I'm not sure anyone has tried to get empirical data before. In any case, 40 degrees is much more believable than my previous guesstimate of about 90 degrees. I was out to lunch on that one. Along with release speed, yaw is almost certainly one of the factors that ultimately determines launch height.

**BELOW** - Yaw of about 40 degrees? 
[image]

PaulZC, 2023-01-16:
I checked the 20948 and it is (or should be!) outputting DPS... Glad your numbers are making more sense!

VancouverUmbrella, 2023-01-16:
Awesome. Thanks again for the superb help, Paul.

PaulZC, 2023-01-18:
Reopening... Just to capture that your 8g issue is resolved with PR #150

PaulZC, 2023-01-18:
Hi Andrew @VancouverUmbrella ,

Please give the updated 2.5 binary a try. Thanks!

Paul

VancouverUmbrella, 2023-01-18:
Definitely will do, Paul.

It's perpetual rain here these days. I'll report back as soon as the skies clear and I can get in a test flight.

With many thanks for your ongoing help,
Andrew

PaulZC, 2023-01-20:
Hi Andrew (@VancouverUmbrella ),

Thanks for the update - glad that's working for you.

64g! Wow! I'm surprised the wings don't fall off! (But I'm guessing these are rather nice carbon-fibre-reinforced beasties?)

Anyway, please reach out again if you need more help.

All the best,
Paul

VancouverUmbrella, 2023-01-21:
Yeah, these gliders are both incredibly strong and very delicate. The little 1m wingspan model I'm using as a testbed can be damaged just by rough handling. However, you can throw it as hard as you're able. Well, the gorilla launchers have been known to break things.

The OLA and KX134 had to go piggyback -- there's just no room inside the fuselage.
[image]