sparkfun/OpenLog_Artemis #148: Feature Request - Please add support for BMA400 - Micro Triple Axis Accelerometer Breakout

skotagiri, 2023-01-15:
### Subject of the issue
Please consider adding support for [BMA400 ](https://www.sparkfun.com/products/21207) Micro Triple Axis Accelerometer Breakout for OLA. I see that the module already contains an Audrino library  at https://github.com/sparkfun/SparkFun_BMA400_Arduino_Library .

I am interested in capturing the data from the build in IMU for OLA as well as a secondary sensor for some sensor fusion experiments. I will attempt to provide a PR if I can get the OLA sketch to build for me locally.

### Expected behavior
When the BMA 400 QWIIC board is plugged in, the OLA log data include the data from the secondary sensor.

### Actual behavior
When the BMA 400 QWIIC board is plugged in, the log data says "unknown device present at address 0x14" and does not log any data from the sensor.

PaulZC, 2023-01-15:
Hello Srikanth (@skotagiri ),

Version 2.5 of the firmware - released yesterday - supports both the [ISM330DHCX](https://www.sparkfun.com/products/20176) IMU and [KX134](https://www.sparkfun.com/products/17589) 64g Accelerometer.

The OLA code is now huge and I'm reluctant to add another accelerometer - unless there is something very unique about it?

Very best wishes,
Paul

srikanthkotagiri-discovery, 2023-01-15:
Thanks for the response. The board is not unique. Just something I had at hand. Thanks for the info on the accelerometers that are already supported.

PaulZC, 2024-02-21:
Closing... I believe the the ISM330DHCX IMU and KX134 64g Accelerometer should meet your needs?

Best wishes,
Paul