sparkfun/OpenLog_Artemis #149: 'class ICM_20948_SPI' has no member named 'initializeDMP'; did you mean 'intEnableDMP'?

userqt, 2023-01-18:
### Subject of the issue
I downloaded the zip for the latest version 2.5 and opened it in Arduino. I installed all required libraries and ran export compiled binary. At some point I get this error:

`'class ICM_20948_SPI' has no member named 'initializeDMP'; did you mean 'intEnableDMP'?`

How should I resolve this? I tried the suggestion intEnableDMP, but that didnt help.

### Expected behavior
Compile completes

### Actual behavior
Process stops due to the error above.

PaulZC, 2023-01-18:
Hi @userqt ,

You need to enable the DMP in the 20948 library. Please follow these instructions:

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/main/DMP.md#is-dmp-support-enabled-by-default

Best wishes,
Paul

userqt, 2023-01-18:
I just tried doing that, but the error still persists.

userqt, 2023-01-18:
ok seems to  be working now. It was the  SparkFun_ICM-20948_ArduinoLibrary that was missing.

PaulZC, 2023-01-18:
Please check your 20948 library is up to date. It should be version 1.2.11.

Please also check that you are compiling this using the SparkFun Apollo3 board definitions (v2.2.1). You need to use **RedBoard Artemis ATP**.

```initializeDMP``` is a _weak_ function. If you are trying to use a different compiler, it may cause problems?

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/f0b90f490762e705441373b6451785e2f9f84a11/src/ICM_20948.h#L237

Please also see these notes. You need to make some small changes in the Apollo3 SPI code:

https://github.com/sparkfun/OpenLog_Artemis/blob/3f0ad73b71bdd257390cd1c7dfe702f4a7caed9f/Firmware/OpenLog_Artemis/OpenLog_Artemis.ino#L113-L123

Compiling the code is difficult. Are sure you cannot use the latest pre-compiled binary? We provide those so users do not need to compile the code themselves.

Best wishes,
Paul

userqt, 2023-01-18:
I wanted to make modifications and add additional libraries, that is why I need to compile it. I was using the RedBoard Artemis. Thank you for the information, I will take a look. I just now noticed another issue with the option 3 missing (seems that online.IMU is false for some reason).

userqt, 2023-01-18:
After I switched to RedBoard Artemis ATP and re-compiled, the option 3 is available. So seems fine. Thanks a lot for your help. I don't know at this point if I need the small changes in the Apollo3 SPI code, things are working fine for now.

PaulZC, 2023-01-18:
You will need the changes if you are going to use the sleep functionality. The code will not sleep or wake correctly without those changes.