sparkfun/OpenLog_Artemis #151: Feature Request: Please add ICM20948 Auto-detection and logging support.

decamun, 2023-01-19:
### Subject of the issue
I'm hoping to use the ICM20948 qwiic boards for my project, because they match the sensor on the artemis itself. However, the firmware currently doesn't automatically detect and make use of these boards. Is it possible to add this functionality?

PaulZC, 2023-01-19:
Hi Devin (@decamun ),

It's a reasonable request but tricky for a number of reasons:
* Adding the SPI code will - I think - take up quite a lot of extra memory (the I2C and SPI versions are separate classes) and the code is already huge
* The 20948 itself is almost end-of-life. We have some stock but it's not certain if we'll be able to get any more
* The firmware now supports the ISM330DHCX IMU, MMC5983MA Magnetometer and KX134 Accelerometer

I'll leave this request open, but it's uncertain if/when we'll be able to implement it.

Best wishes,
Paul

decamun, 2023-01-19:
Hey Paul, 

Thanks for letting me know. That's understandable. I was beginning to work on this myself, but I suppose it might be easier to simply return the ICM20948 sensors and get some ISM330DHCX's instead. 

Best,
Devin

PaulZC, 2024-02-21:
Closing... Sorry! It's just not practicable to support this request when the firmware already supports the ISM330DHCX IMU. Please also check out the [DataLogger IoT](https://www.sparkfun.com/products/20594).

Best wishes,
Paul

closed 2024-02-21 by PaulZC