sparkfun/OpenLog_Artemis #158: Feature - documentation for building binaries

jliphard, 2023-02-17:
The prebuilt binaries are very useful, but might it be possible to provide instructions for building new binaries? For example, for a Red OpenLog Artemis, what are the correct `defines` and correct settings for the Arduino IDE? 

* Setting the Arduino IDE to `Redboard Artemis Nano` - is that right?
* And:
```
//#define noPowerLossProtection
#define HARDWARE_VERSION_MAJOR 1
#define HARDWARE_VERSION_MINOR 0
#define SD_FAT_TYPE 3
```
When I generate the binary that way, it compiles (after installing many libraries), but it does not run. By contrast, `OpenLog_Artemis-V10-v25.bin` works immediately. Anything obvious I'm doing wrong?

Also, despite the assertion in the comments, the boards do not seem to have a revision number?

PaulZC, 2023-02-17:
Hi Jan (@jliphard ),

Building the binary is not easy. But, if you're up for a challenge, I've added abbreviated instructions here:

https://github.com/sparkfun/OpenLog_Artemis/blob/main/COMPILE_BINARY.md

If you don't mind, I'll use you as a Beta-tester. Please let me know if the instructions are clear enough and if you get the code to compile.

Thank you,
Paul

jliphard, 2023-02-17:
Thanks so much! Minor running notes:

1. Step 2 - path is actually “main" not “master”? In `https://raw.githubusercontent.com/sparkfun/Arduino_Apollo3/master/package_sparkfun_apollo3_index.json`
2. "Copy Paulvha's Patch for Apollo3" Confusing wording - basically, just replace five files in `Arduino15\packages\SparkFun\hardware\apollo3\2.2.1` with five patched files from the .zip?
3. Could the "extra" file just be added to the main repo?

Otherwise, **very straightforward and worked first try**. 

I'm using using: 
Arduino Version: 2.0.4-nightly-20230215
Mac ARM Silicon

PaulZC, 2023-02-18:
Hi Jan (@jliphard ),

Thanks for the feedback - much appreciated. Glad that's working for you - especially on Mac and with IDE v2!

1) Yes, indeed. Changed... The branch name was changed months ago - but the old name does still work.
2) Changed - to hopefully less confusing wording.
3) The core libmbed-os itself needs patching... That's tricky - and above my pay grade. Paulvha kindly provided a work-around with the pre-compiled .a binary for the ATP.

I'll close this, but please re-open if you need further help.

Best wishes,
Paul

closed 2023-02-18 by PaulZC