sparkfun/OpenLog_Artemis #166: Flush input buffers on wake

whipple63, 2023-02-28:
We have an external BLE module that we power from the I2C bus so that it can be shut down between samples.  It communicates through serial1 (Rx/Tx pins).  When it wakes up it generates noise on the communication lines that the OLA interprets as having sent a character which sends the OLA into the Main Menu.  An attempt to address this was made using the option in the debug menu to only open the menu on a printable character, however recently the noise is apparently being interpreted as a printable character.  This issue is solved by flushing the input buffer(s) near the end of wakeFromSleep().  Doing this near the end allows time for the module to power up and generate noise before flushing the buffer.
The question is then, is there any reason not to flush the input buffer(s) on wake up?  The code is simple just before the end of wakeFromSleep():

  // Late in the process to allow time for external device to generate unwanted signals
  while(Serial.available())  // Flush the input buffer
    Serial.read();
  if (settings.useTxRxPinsForTerminal == true)
  {
    while(Serial1.available())  // Flush the input buffer
      Serial1.read();
  }
  
  //When we wake up micros has been reset to zero so we need to let the main loop know to take a reading
  takeReading = true;
}

And again, happy to do a pull-request if this is a good change.

PaulZC, 2023-03-02:
Hi Tony,

This looks like another good change. A PR is always good, then you get full credit! ;-)

I'm neck deep in another product at the moment, so it may be a while before I can merge your changes and generate the binaries.... Please don't think I'm ignoring you... We appreciate the updates.

Best wishes,
Paul

PaulZC, 2024-02-21:
This too is included in v2.6. Thanks Tony!

closed 2024-02-26 by PaulZC