sparkfun/OpenLog_Artemis #171: Issue - serial communication error with one non-standard "byte"

Gerard-E, 2023-05-08:
Hello,

Currently I'm working on a project where I need to log a serial protocol.
I'm talking about the LIN protocol, which is very similar to the UART protocol.

Similarities:
- Idle high
- Compatible baud rate 19k2
- 8N1 aka: low start bit, 8 data bits, high stop bit

Difference:
The first "byte" of a message is called a Break field, and consists of a low signal with a duration of 13 nominal bits times.
This Break field does not comply with the 8N1 serial standard.
_I've added a screenshot of the Break field specification at the end of this message._

The problem:
An FTDI cable to a PC as well as an Arduino Uno or Mega will read this Break field as 0x00.
The Artemis openlog does not seem to read this break field , but I do need it in my logging file.

Can this problem be solved in code or is this a limitation of the Appolo 3 chip?

[image]
Source: https://lin-cia.org/fileadmin/microsites/lin-cia.org/resources/documents/LIN-Spec_Pac2_1.pdf

PaulZC, 2023-05-08:
Hello @Gerard-E ,

I do not know how the MBED core and Apollo3 hardware responds to the break character. It may be possible to change the core so it records a NULL (0x00) like the Uno/Mega, but it would involve modifying the core at a very low level.

Does the OpenLog Artemis record any data from your LIN bus / device? Do you see valid data in the log file - except that the breaks are missing?

Could you use a Mega to convert the break field into an actual 0x00 byte - which the OLA can record? Connect your LIN bus / device to Serial1 RX. Connect Serial1 TX to the OLA. Be careful with the voltages - the original Arduino Mega is a 5V board. Use something like:

```
void setup()
{
  Serial1.begin(19200);
}

void loop()
{
  while (Serial1.available())
    Serial1.write(Serial1.read());
}
```

Best wishes,
Paul

PaulZC, 2023-06-27:
Closing. Please reopen if you need more help or advice with this issue.

Best wishes,
Paul

closed 2023-06-27 by PaulZC