sparkfun/OpenLog_Artemis #172: Provide possibility to log in binary format for some of the sensors

jerabaul29, 2023-06-27:
Would it be possible to provide possibilities to log in binary format, at least for some of the sensors supported? The reason is that this would be much more space effective, which means fewer writes (which can be a bit power intensive), need for less big SD cards, etc.

For example, in our case with @larswd- , we would like to log the ISM330DHCX at high frequency (ideally 800Hz or higher), for long periods of time (weeks to start with, but going to months would be great). This means logging 6 float32s (accX, accY, accZ, gyrX, gyrY, gyrZ9 at 800Hz. Using binary data, + 1-char "start message header", + timestamp in uint32_t, + 1-char "newline" at the end of each entry, this could be in theory reduced to:

6 * 4 + 1 + 4 + 1 = 30 bytes

by logging all of this with chars / ascii, we will need typically at least on the order of 100 bytes per record I think? So the savings are not negligible.

Could adding such a "binary logging format" on a sensor to sensor basis for sensors that potentially work at high frequency be included as an additional option in the firmware?

PaulZC, 2023-06-27:
Hi JR,

Sadly there is no way to achieve 800Hz with the ISM330DHCX on I2C on OpenLog Artemis. I suspect you'll be limited to something like 100-200Hz at best, and only if you strip everything to the bone...

For this application, I'd recommend using the new [DataLogger IoT 9DoF](https://www.sparkfun.com/products/20594). The IMU and Mag are connected via SPI and both can use interrupts. The microSD card is connected via 4-bit SDIO, which is an order of magnitude faster than SPI. There is an [example](https://github.com/sparkfun/SparkFun_DataLogger_IoT_9DoF/blob/main/Examples/Hardware_Tests/12_Log_Fast_IMU/12_Log_Fast_IMU.ino) which will get you started. However, please see the big warning in the example code. If you overwrite the DataLogger firmware with the code example, you can not put it back on again.

Very best wishes,
Paul

closed 2023-06-27 by PaulZC

jerabaul29, 2023-06-27:
@PaulZC that sounds very strange - I do get 800Hz with the Ambiq + ISM330DHCX on the OpenMetBuoy.

See the project's code as a whole at: https://github.com/jerabaul29/OpenMetBuoy-v2021a/tree/main/legacy_firmware/firmware/standard_gps_waves_drifter . 

I think the trick to get it to work at high frequency is "simply" to set up the speed / clock frequency of the I2C port:

https://github.com/jerabaul29/OpenMetBuoy-v2021a/blob/430ad44ec9ee7e0cec8927b25ca8eaea1b008501/legacy_firmware/firmware/standard_gps_waves_drifter/sleep_manager.cpp#L138-L143

With this in place, I can positively confirm that it is possible to log the ISM330DHCX with this kind of settings:

https://github.com/jerabaul29/OpenMetBuoy-v2021a/blob/430ad44ec9ee7e0cec8927b25ca8eaea1b008501/legacy_firmware/firmware/standard_gps_waves_drifter/imu_manager.cpp#L96-L101

(the OMB has been validated many times by several groups separately in laboratory calibration tests, so I am very confident it works as it should :) ).

Would it be possible then to provide an additional setting to be able to set the I2C port frequency higher in case some logging tasks require it? :) I do not think this comes at any costs in the case of the OMB, though I suppose some other sensors may not support these higher clock frequencies?

jerabaul29, 2023-06-27:
Can you re-open if this makes sense to you (and let me know if I should open an additional issue about "providing an option to select the I2C port frequency :) ).

PaulZC, 2023-06-27:
Hi JR,

Currently the OLA supports 100kHz and 400kHz I2C. I see you are running the OMB bus at 1MHz. OK. You are welcome to open a 1MHz I2C Feature Request and I will get to it when time permits.

The binary format is too big a change for me to consider adding it. It completely violates the data files being human-readable CSV. It would be a big code change and again very different to how most users use the OLA. And there may be hidden 'gotchas' in there, like no longer being able to use strlen to calculate the length of each line. I still think you would get much better results writing your own dedicated firmware for a board more suited to your needs.

Best wishes,
Paul