sparkfun/OpenLog_Artemis #174: Feature Request: Separate logging frequencies for different sensors

larswd, 2023-06-30:
### Subject of the issue
Me and @jerabaul29 desire to use the OpenLog Artemis with the [ISM330DHCX](https://www.sparkfun.com/products/19764) and [MAX-M10S GNSS Receiver](https://www.sparkfun.com/products/18037), and as mentioned in issue #172 we need a high frequency on the IMU logging capability.  As we plan to make many sensors with the same components, we will need the GPS to both know the location and  ensure that each sensor has an accurate time stamp available. Yet, we also do not need to do these measurements frequently and i would therefore be nice if we could be able to set different logging frequencies for the IMU and the GPS to preserve power and reduce the amount of unnecessary data being stored. Could this be done?

PaulZC, 2023-06-30:
Hi Lars (@larswd ),

No, sorry, that's essentially impossible with the existing firmware. The firmware is designed to sample all connected sensors, each in turn. Adding different "decimation rates" for each sensor type would be a huge change. I would encourage you to write your own firmware for the OLA, optimized to your individual needs.

Best wishes,
Paul

closed 2023-06-30 by PaulZC

jerabaul29, 2023-06-30:
@PaulZC does this mean that one cannot log the GPS at 1Hz and the ISM330DHCX at 100HZ? Any idea of a workaround? Or is it "just fine" to log both at 100Hz, even if the GPS is actually updated at 10Hz only?

PaulZC, 2023-06-30:
You can set the GNSS to "auto". It will then report the most recent 'stale' fix if no fresh data is available. You can access this option through the "Detect / Configure Attached Devices" menu.

If you set the logging rate to 100Hz, the GNSS will be configured at the maximum rate of 10Hz.

jerabaul29, 2023-06-30:
Ok, thanks for the explanation :) . Is it already what you do @larswd ? :)

larswd, 2023-07-03:
Sorry for the late reply! Yes, that is the solution I have gone with thus far, and the only big drawback seems to be that we still are limitied in write speed, so we might have to pick and choose a bit in what we log. I will open a separate issue on higher baud rate to complement your request for a higher I^2C frequency request.