sparkfun/OpenLog_Artemis #175: Feature request. Higher baud rate for logging to SD card

larswd, 2023-07-03:
Following up on the discussion in #174 and in the same spirit as #173, @jerabaul29 would like to request for a possible lift on the Serial baud rate cap, as we have not been able to aquire a logging frequency of more than 250 Hz, which falls a bit short of our desired target. Would an increase to the maximal baud rate of 500 000 be possible?

jerabaul29, 2023-07-03:
It should be possible to get it up to 1M without issues :) .

PaulZC, 2023-07-03:
Hi Lars (@larswd ),

Have you tried disabling the "Log to Terminal" option - from the "Configure Terminal Output" menu? That helps to increase the logging rate. The sensor data is still logged to file.

Best wishes,
Paul

larswd, 2023-07-04:
Yes, although I at least did not see it give a significant change in
logging frequency when inspecting the log files. Maybe I misremember, but I
am certain it did not give the necessary boost in logging frequency. But I
will test that some more when I am back home. It is a good suggestion!

PaulZC, 2024-05-23:
Closing... Please reopen if you need more suggestions.

Best wishes,
Paul