sparkfun/OpenLog_Artemis #176: Feature Request/Question: Cannot get OpenLog Artemis to log from `loadcell-nau7802`

gmkado, 2023-08-23:
### Subject of the issue
I can't get my loadcell to work with openlog artemis.  Not sure if its due to the load cell reporting negative raw values, but I'm just getting empty entries (not zeros in compression + values in tension like I would expect if that were the case).  Read on for details

### Your workbench
* Are you using a microSD card? If so, what size? How is it formatted (FAT, FAT32, or exFat)?
No microSD
* At what frequency are you logging? For example, 10Hz, 1Hz, once every 5 seconds, etc.
Tried 10Hz and 1Hz
* What version of firmware are you using? This is shown at boot, for example: "Artemis OpenLog v1.10"
v2.4

* How is OpenLog Artemis wired to your sensor(s)?
I have the following connections:
    - OLA hooked up to my laptop via USB-C
    - OLA to SEN board via qwiic connector
    - Wires from SEN board to load cell using color code:
Input: Red wire(+), Black wire(-)
Output: Green wire(+), White wire(-)

* How is everything being powered?
Via the usb-c from my laptop

* Are there any additional details that may help us help you?
My shopping cart:
- [1568-SEN-15242-ND](https://www.digikey.com/en/products/detail/sparkfun-electronics/SEN-15242/10064419)
- [1568-DEV-16832-ND](https://www.digikey.com/en/products/detail/sparkfun-electronics/DEV-16832/13166206)
- https://www.amazon.com/dp/B077YHNNCP?psc=1&ref=ppx_yo2ov_dt_b_product_details

### Steps to reproduce

Make all connections and open serial port using arduino IDE.  

One strange thing to note, initially in the weight field I was getting weird characters like ⸮.  So I initially thought it was a text encoding issue, (but everything else was printing out fine).  So I tried another serial port monitor that allowed me to change the encoding to e.g. UTF-8, 16, etc but nothing there worked.

After that I tried removing datalogging, and I think at that point it stopped printing out weird characters and just prints out empty entries like in the snippet below.  So I thought it might be a rate issue and slowed the polling down to 1Hz.  I also tried the baud rate at 115200 and 9600, made no difference.

### Expected behavior
Values are logged in the "weight" column of the output

### Actual behavior
This is what's logged:
```
SD init failed (second attempt). Is card present? Formatted?
Please ensure the SD card is formatted correctly using https://www.sdcard.org/downloads/formatter/
Config file read failed: SD offline
Artemis OpenLog v2.4
SD card offline
Datalogging offline
Serial logging offline
IMU offline - or not present
Identifying Qwiic Muxes...
Identifying Qwiic Devices...
Autodetect complete
Device config file read failed: SD offline
LoadCell-NAU7802 online at address 0x2A
Content-Type: text/csv

rtcDate,rtcTime,weight(no unit),output_Hz,
01/01/2000,02:29:14.04,,100.0,
01/01/2000,02:29:15.05,,1.961,
01/01/2000,02:29:16.05,,1.485,
01/01/2000,02:29:17.06,,1.320,
01/01/2000,02:29:18.06,,1.241,
01/01/2000,02:29:19.07,,1.190,
01/01/2000,02:29:20.07,,1.159,
01/01/2000,02:29:21.08,,1.135,
```

I also ran [this example](https://github.com/sparkfun/SparkFun_Qwiic_Scale_NAU7802_Arduino_Library/tree/master/examples/Example3_Gain_SampleRate) off a spare arduino and am able to see the values update.  

This led me to discover I was getting zeros for "compression" and actual values for "tension", which led me to this [forum post](https://forum.sparkfun.com/viewtopic.php?t=55611), which led me here.

Following the advice on that post, I changed the "allowNegativeWeights" to true in the sketch, and I get the following output:
```
Reading: -190173	Weight: 1.73	AvgWeight: 1.71	Scale not calibrated. Press 'c'.
Reading: -192422	Weight: 1.71	AvgWeight: 1.71	Scale not calibrated. Press 'c'.
Reading: -188475	Weight: 1.70	AvgWeight: 1.72	Scale not calibrated. Press 'c'.
Reading: -184800	Weight: 1.68	AvgWeight: 1.71	Scale not calibrated. Press 'c'.
Reading: -183763	Weight: 1.64	AvgWeight: 1.68	Scale not calibrated. Press 'c'.
Reading: -179378	Weight: 1.59	AvgWeight: 1.65	Scale not calibrated. Press 'c'.
Reading: -172773	Weight: 1.57	AvgWeight: 1.62	Scale not calibrated. Press 'c'.
Reading: -172112	Weight: 1.55	AvgWeight: 1.59	Scale not calibrated. Press 'c'.
```
(Side note, I could not get the scale calibration to "pass", it always says scale not calibrated...)

So I went back to the OLA and thought maybe it would log data if I kept it in tension (aka positive values) but that doesn't seem to work.  

Would appreciate any help getting this thing to log.  Thanks!

PaulZC, 2023-08-23:
Hi @gmkado ,

You need to calibrate the NAU7802 through the OpenLog Artemis menus.
Open the serial Menu and select option 6 "Detect / Configure Attached Devices".
Select the NAU7802 from the list.
From the NAU7802 sub-menu, select option 2 "Calibrate Scale".
It will guide you through setting the zero point and then calibrating with a known weight.

Unfortunately you need to do this each time the OLA restarts. The firmware is not able to save the zero point and calibration factor - as this depends on whether the NAU7802 is connected through a mux. The [DataLogger IoT](https://www.sparkfun.com/products/20594) is able to save the configuration - but currently does not support muxes...

Let me know if you need more help.

Best wishes,
Paul

PaulZC, 2023-08-23:
OK. I just saw your Forum post:

https://forum.sparkfun.com/viewtopic.php?p=243322#p243322

I will change this to a "negative weight" feature request.

gmkado, 2023-08-26:
@PaulZC thanks for the help!  I just tried what you suggested but I can't get past the sensor menu:
[image]

Pressing `1` and `enter` doesn't do anything.  Any idea why?

Also regarding this statement:
> Unfortunately you need to do this each time the OLA restarts. The firmware is not able to save the zero point and calibration factor - as this depends on whether the NAU7802 is connected through a [...]

Does this mean I can save calibration If I'm not using a mux?  Also, there's no way to just log the raw values?

PaulZC, 2023-08-27:
Hi,

For the sensor menu: check your line endings (how you have the enter key configured). If you are using the Arduino IDE Serial Monitor, set the line ending to Newline. If you are using a terminal emulator, dig around in the settings for new-line. You need to send a New Line '\n' (ASCII 10 (decimal)) - or Carriage Return '\r' (ASCII 13 (decimal)) - or both. (Some menus respond to a single character - the '1' on its own. Some need the terminator. That menu needs the terminator.)

Sorry, no, you cannot save the calibration at all at present. To do it properly, the firmware would need to record the mux address and port (if connected), or NULL values if connected on the main I2C bus. That requires new code... You could 'cheat' and record the zero point and calibration factor for _only_ a single device on the main bus. That would be much easier but again still requires new code. Sorry, no, it is not possible to record raw values.

Best wishes,
Paul

gmkado, 2023-08-28:
Okay so it seems like the OLA in its current implementation won't work for my application since it requires some user interaction on power cycle.

I'm going to switch to using the [pro-micro ](https://www.sparkfun.com/products/15795) along with the [arduino library](https://github.com/sparkfun/SparkFun_Qwiic_Scale_NAU7802_Arduino_Library/tree/master).  You can close this ticket

PaulZC, 2024-02-26:
Hi @gmkado ,

It may be too little too late, but this issue is resolved in firmware v2.6. I found a bug which was preventing the NAU7802 from being configured correctly when connected through a mux. This is now resolved. And you can log negative weights too. The configuration for each device is stored on SD Card and is now restored correctly on power-up or wake. Please check the v2.6 release notes for more details.

Best wishes,
Paul

closed 2024-02-26 by PaulZC