sparkfun/OpenLog_Artemis #180: Timeline for BLE?

drewhamiltonasdf, 2023-09-27:
Hi,

I noticed in the Hookup Guide (https://learn.sparkfun.com/tutorials/openlog-artemis-hookup-guide/all) there is mention of BLE capability in the "coming months" so I headed over to look at the BLE branch and I see that the last commit was about 2 years ago. 

Certainly not complaining, but just curious if there is any timeline for that feature release? I am very much in need of this and weighing the pros and cons of trying to "get it to work" on my own, but would much prefer waiting if the release is on the horizon.

Many thanks,
Drew

PaulZC, 2023-09-27:
Hi Drew,

Unfortunately we do not plan to add BLE capability on OpenLog Artemis. The Artemis is BLE-capable and [other customers](https://github.com/sparkfun/OpenLog_Artemis/issues/66) have been successful getting BLE up and running. But we don't have the resources to formally add support for it in the OLA firmware.

Apologies,
Paul

drewhamiltonasdf, 2023-11-02:
If anyone runs across this. I found it relatively straight-forward to implement BLE using the pointers found in the link provided by @PaulZC . It is not integrated into the whole ArtemisOpenlog Firmware system, but it could be with a little elbow grease. I'll close this issue for now.

closed 2023-11-02 by drewhamiltonasdf