sparkfun/OpenLog_Artemis #182: Trouble with logging serial data

HolgerSiebert, 2023-11-14:
### Subject of the issue
Hello, I want to log some serial ascii data at baudrate of 115200 including a time stamp. When the data source is already running before the OLA has been started, OLA is crashing with the following message: 

 ++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: handle != NULL
Location: 0x49DBD
File: mbed-os/targets/TARGET_Ambiq_Micro/TARGET_Apollo3/device/serial_api.c+411
Error Value: 0x0
Current Thread: main Id: 0x10007E4C Entry: 0x4AF11 StackSize: 0x1000 StackMem: 0x10006E28 SP: 0x1005FEEC
For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt=SFE_ARTEMIS_ATP
-- MbedOS Error Info --

If I start the data source after OLA is already running and storing data, it works more or less reliable (with a few other issues I will come back later)

### Your workbench
* SC Card is a " 32GB extreme pro"  formatted with exFAT (1024 kb units)
* I log data at 50 Hz
* firmware is v2.4
* How is OpenLog Artemis wired to your sensor(s)? serial data source is wired via a MAX232 (RS232 =>logic) at OLA PIN RX, MAX232 is powered by OLA (3.3V), other devieces via 
* How is everything being powered? OLA is powered via USB-C; serial decive is powerd by USB, all on common GND
* Are there any additional details that may help us help you? Other QwiicDevices are: 

### Steps to reproduce
sizeOfSettings=168
olaIdentifier=292
nextSerialLogNumber=0
nextDataLogNumber=0
usBetweenReadings=20000
logMaxRate=0
enableRTC=1
enableIMU=0
enableTerminalOutput=0
logDate=1
logTime=1
logData=1
logSerial=1
logIMUAccel=0
logIMUGyro=0
logIMUMag=0
logIMUTemp=0
logRTC=1
logHertz=0
correctForDST=0
dateStyle=0
hour24Style=1
serialTerminalBaudRate=115200
serialLogBaudRate=115200
showHelperText=1
logA11=0
logA12=1
logA13=0
logA32=1
logAnalogVoltages=1
localUTCOffset=0.00
printDebugMessages=1
powerDownQwiicBusBetweenReads=0
qwiicBusMaxSpeed=100000
qwiicBusPowerUpDelayMs=250
printMeasurementCount=1
enablePwrLedDuringSleep=1
logVIN=0
openNewLogFilesAfter=10
vinCorrectionFactor=1.47
useGPIO32ForStopLogging=0
qwiicBusPullUps=1
outputSerial=0
zmodemStartDelay=5
enableLowBatteryDetection=0
lowBatteryThreshold=3.40
frequentFileAccessTimestamps=0
useGPIO11ForTrigger=0
fallingEdgeTrigger=1
imuAccDLPF=0
imuGyroDLPF=0
imuAccFSS=0
imuAccDLPFBW=7
imuGyroFSS=0
imuGyroDLPFBW=7
logMicroseconds=0
useTxRxPinsForTerminal=0
timestampSerial=1
timeStampToken=10
useGPIO11ForFastSlowLogging=0
slowLoggingWhenPin11Is=0
useRTCForFastSlowLogging=0
slowLoggingIntervalSeconds=300
slowLoggingStartMOD=1260
slowLoggingStopMOD=420
resetOnZeroDeviceCount=0
imuUseDMP=0
imuLogDMPQuat6=1
imuLogDMPQuat9=0
imuLogDMPAccel=0
imuLogDMPGyro=0
imuLogDMPCpass=0
minimumAwakeTimeMillis=0
identifyBioSensorHubs=0
serialTxRxDuringSleep=0
printGNSSDebugMessages=0
openMenuWithPrintable=0

### Expected behavior
Even if serial device is already running when powering up OLA (or rest OLA) I expect data to be logged 

### Actual behavior

OLA does not start logging but stops with the following message: 

 ++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: handle != NULL
Location: 0x49DBD
File: mbed-os/targets/TARGET_Ambiq_Micro/TARGET_Apollo3/device/serial_api.c+411
Error Value: 0x0
Current Thread: main Id: 0x10007E4C Entry: 0x4AF11 StackSize: 0x1000 StackMem: 0x10006E28 SP: 0x1005FEEC
For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt=SFE_ARTEMIS_ATP
-- MbedOS Error Info --

PaulZC, 2023-11-15:
Hi @HolgerSiebert ,

It's not clear to me if this is an issue with the Apollo3, or the OLA firmware, or perhaps a strange power issue from the way you have the MAX232 connected. I will try some tests, but it will be difficult for me to fully replicate your set-up.

Question: are you using a MAX232, or a MAX3232? The MAX232 is, IIRC, a 5V-only part. The MAX3232 can operate from +3.0V to +5.5V.

Best wishes,
Paul

PaulZC, 2024-02-19:
Closing... Please re-open if you need more help with this issue.

Best wishes,
Paul

PaulZC, 2024-03-04:
Reopening...

Another user reported the same issue here: https://forum.sparkfun.com/viewtopic.php?p=247614#p247614

The root cause of the crash is serial data arriving after the ```Serial1.end``` [here](https://github.com/sparkfun/OpenLog_Artemis/blob/8ad4b4046e08621908869a5034efe38322bf789d/Firmware/OpenLog_Artemis/OpenLog_Artemis.ino#L471), and before ```Serial1``` is restarted by ```beginSerialLogging``` [here](https://github.com/sparkfun/OpenLog_Artemis/blob/8ad4b4046e08621908869a5034efe38322bf789d/Firmware/OpenLog_Artemis/OpenLog_Artemis.ino#L525).

v2.7_BETA will be available shortly.

PaulZC, 2024-03-04:
Please try the 2.7 BETA firmware available [here](https://github.com/sparkfun/OpenLog_Artemis/tree/main/Binaries) and [here](https://github.com/sparkfun/OpenLog_Artemis/blob/main/Binaries/OpenLog_Artemis-V10-v27_BETA.bin)

Upgrade instructions are [here](https://github.com/sparkfun/OpenLog_Artemis/blob/main/UPGRADE.md)

PaulZC, 2024-03-05:
Hi @HolgerSiebert ,

Apologies for the slow response. This is resolved in [v2.7](https://github.com/sparkfun/OpenLog_Artemis/releases/tag/v2.7).

Best wishes,
Paul

closed 2024-03-05 by PaulZC