sparkfun/OpenLog_Artemis #187: Feature request: sensor value triggers analog outputs

n8maher, 2024-02-23:
How cool would it be to be able to set a sensor value at which the analog output would be triggered? For example, lighting up an LED when a certain weight was reached on a scale, or to control a fan based on a temperature reading. 

In the sensor setup, you would need to add a trigger value, and then set whether to trigger when above or below this value. Bonus points would be given if the trigger could be based on the last x readings or if a certain amount of 'hysteresis' would be required in order for the trigger to be canceled.

Cheers, all!

n8maher, 2024-02-23:
If this gets traction, the next step in this direction would be for the Openlog Artemis to recognize the Qwiic motor driver, GPIO board, and Quad relay boards to provide even more ability to control things based on sensor values.

PaulZC, 2024-02-24:
Hi @n8maher ,

It's a good suggestion... But this needs a ton of work to do it properly. Each sensor could have multiple things which could be triggers. Some would be int, some float. The menus would grow exponentially. And the loop code would need a list of registered / active triggers and would need to check each one in turn...

Like I say, a good suggestion, but just not practicable on OLA.

This _could_ be a good feature for the [DataLogger](https://github.com/sparkfun/SparkFun_DataLogger) though. The DataLogger firmware is much more flexible and much better structured. Adding trigger values there should be possible. Please consider opening a Feature Request over there. Thanks!

Best wishes,
Paul

closed 2024-02-24 by PaulZC