sparkfun/OpenLog_Artemis #192: time stamps in serial data file

HolgerSiebert, 2024-03-07:
### time stamps in serial data file

Dear Paul

thanks for solving issue #182 but one issue remains with logging serial data (same setupas in #182 but now with firmware 2.7 beta): I want to include a timestamp in the serial data file which looks like (without time stamp)

S  01.54 D  087 U -01.53 V -00.09 W  00.14 T  09.88 H  62.07 P  1009.21 PI  000.4 RO -006.2 MD  348
S  01.31 D  082 U -01.29 V -00.18 W  00.19 T  09.55 H  63.58 P  1009.23 PI  000.4 RO -006.2 MD  359
S  01.29 D  087 U -01.29 V -00.06 W  00.03 T  09.43 H  64.10 P  1009.24 PI  000.4 RO -006.2 MD  348

I asked OLA to include the time stamp after token = 84 ("T") which is unique in every individual telegram; however, the time stamp appears  "somewhere" and my logged data looks like: 

 
S  01.54 D  087 U -01.53 V -00.09 W  00.14 T  09.88 H  62
^02/28/2024,12:28:12.63
.07 P  1009.21 PI  000.4 RO -006.2 MD  348
S  01.31 D  082 U -01.29 V -00.18 W  00.19 T  09.55 H  63
^02/28/2024,12:28:12.67
.58 P  1009.23 PI  000.4 RO -006.2 MD  359
S  01.29 D  087 U -01.29 V -00.06 W  00.03 T  09.43 H  64
^02/28/2024,12:28:12.69
.10 P  1009.24 PI  000.4 RO -006.2 MD  348
 
 
Sometimes the data looks even worse and parts of the telegram are missing; I never observed this issue when logging without time stamps so I assume the general serial data transmission works fine (after you solved issue #182). I also tried to add the time stamp after the carriage return/line feed at the end of the telegram but with the same result

Any thoughts about this?

Best wishes 
Holger

PaulZC, 2024-03-07:
Hi Holger (@HolgerSiebert ),

Please try v2.8_BETA : https://github.com/sparkfun/OpenLog_Artemis/tree/main/Binaries

Best wishes,
Paul

HolgerSiebert, 2024-03-07:
Hi Paul,

thanks for your prompt response - I only had a first look at the results but it looks promising. I will report in detail later.

Best wishes

Holger

--------
Holger Siebert PhD
Leibniz Institute for Tropospheric Research (TROPOS)
Dept of Experimental Aerosol and Cloud Microphysics

Permoserstr 15
04318 Leipzig

Phone: ++49 341 2717 7145
Email: ***@***.***

Hi Holger ***@***.***<https://github.com/HolgerSiebert> ),

PaulZC, 2024-05-23:
Closing... Please reopen if you need more help with this.

commit 86d6c109 (2024-03-07): v2.8 BETA - to resolve #192
commit 81445bd4 (2024-03-07): Merge pull request #193 from sparkfun/release_candidate

v2.8 BETA - to resolve #192
closed 2024-05-23 by PaulZC