sparkfun/OpenLog_Artemis #194: OLA used as a Blackbox for INAV Drones

Hex-Davis, 2024-04-30:
### Subject of the issue
The data collected by OLA via seral logging is not readable by 'INAV blackbox explorer'. I can bring it up in 'Notepad++' and read headers but the actual data seems to be in a program language or something, and is not registering with the blackbox explorere

### Your workbench
* MicroSD card: 32G Fat32 format
* At what frequency are you logging? 5Khz, all onboard sensors are disabled and only the serial data is actual being captured.
* Version of firmware: Artemis OpenLog v2.8
* How is OpenLog Artemis wired to your sensor(s)?: via the Rx pin 
* How is everything being powered?: via 5v on board regulator on the flight controller.
* Additional details:

### Steps to reproduce
1) Verify drone with INAV flight software has the ports tab set with the correct UART peripheral set to Blackbox then ensure the baud rate on that UART's peripheral matches the serial baud rate in the OLA menus.
2) Power up the drone and push the reset button on the OLA.
3) Once the lights on the OLA are flashing ARM the drone and the OLA will start collecting data. 
4) After the flight upload data to the 'INAV blackbox explorer' ensuring to use the pin 32 stop logging to save the data before pulling power.

### Expected behavior
The data collected during the flight, should give you a break down of all the onboard sensors reading as the quad was flying second by second.

### Actual behavior
The data collected appears to  not be read and the explorer only has a single point of data for the flight, despite the very large amount of data collected.

**Any help is greatly appreciated** 

### Sample DATA
[# Diff all.txt](https://github.com/sparkfun/OpenLog_Artemis/files/15156252/Diff.all.txt)
[# dump.txt](https://github.com/sparkfun/OpenLog_Artemis/files/15156253/dump.txt)
[Data0001_Baud115200_BBR_1over8_Looptime125.TXT](https://github.com/sparkfun/OpenLog_Artemis/files/15156255/Data0001_Baud115200_BBR_1over8_Looptime125.TXT)
[Data0004_Baud57600_BBR_1over8_Looptime125.TXT](https://github.com/sparkfun/OpenLog_Artemis/files/15156257/Data0004_Baud57600_BBR_1over8_Looptime125.TXT)
[Data0006_Baud57600_BBR_1over32_Looptime125.TXT](https://github.com/sparkfun/OpenLog_Artemis/files/15156259/Data0006_Baud57600_BBR_1over32_Looptime125.TXT)

PaulZC, 2024-04-30:
Hi @Hex-Davis ,

I don't think we (SparkFun) can give you much help with this. But I'm happy to leave this issue open in case anyone else can.

Best wishes,
Paul

Hex-Davis, 2024-05-01:
I kind of figured it was a long shot. I think its just a difference in languages used but I don't have any formal background in programing so I'm lost. 

Thanks for leaving this issue open, any help will be appreciated.

muhammedoney, 2025-02-24:
I have the same issue 👍 Can you figure it out the solution