sparkfun/OpenLog_Artemis #203: Unable to add Library to Arduino IDE 2.3.3 or 1.8.12 - Error: 13 INTERNAL - LIBRARY NOT VALID

plema21, 2024-11-13:
### Error adding OpenLog_Artemis to Arduino IDE

Hello, I am looking to incorporate the OLA into a RedBoard/Arduino Project. When trying to install the associated OpenLog_Artemis Library I am confronted with an error from Arduino IDE: (Error: 13 INTERNAL: Library install failed: moving extracted archive to destination dir: - LIBRARY NOT VALID).

I have tried on Ardiuno IDE 2.3.3 and 1.8.12 on two different machines. I am attempting to add the library via Sketch-> Include Library -> Add .ZIP Library. I am unsure how to interface with the OLA via Arduino without this library. I am attempting to log very basic processed data acquired through qwiic sensors and manipulated in Arduino IDE. Please excuse any ignorance in this issue submission. 

### Your workbench
OLA works as expected when connected to various QWIIC sensors, but I have no control over this module via Arduino IDE. I would like to collect atmospheric data and combine with differential pressure data to calculate volumetric airflow and store this value to the OLA.

### Steps to reproduce
Try adding this library via Arduino IDE as mentioned above :   Add library via Sketch-> Include Library -> Add .ZIP Library

### Expected behavior
Library successfully added

### Actual behavior
(Error: 13 INTERNAL: Library install failed: moving extracted archive to destination dir: - LIBRARY NOT VALID).

PaulZC, 2024-11-13:
Hi @plema21 ,

The OLA firmware isn't a Library. It is a complex "sketch" which incorporates (includes) all of the individual sensor libraries.

You can use the OLA firmware to store the raw sensor values, and then post-process the CSV file to calculate your airflow.

Or - a _much_ larger project - you could adapt the OLA firmware so it can calculate the airflow in real time. You can find compilation instructions [here](https://github.com/sparkfun/OpenLog_Artemis/blob/main/COMPILE_BINARY.md).

I hope this helps.

Best wishes,
Paul

closed 2024-11-13 by PaulZC

plema21, 2024-11-13:
PaulZC, 

Thank you so much for this explanation! You have saved me countless time trying to figure this out.

Best ,

A fellow Paul