sparkfun/OpenLog_Artemis #208: data logging limitations on serial log - possible workaround?

Beepnblink, 2025-02-03:
Hi Paul et al,

I have seen issue #201 reporting data corruption, but maybe this issue is different.
We are logging serial data at 38400Bd every 60s. So not likely a problem where the OLA can't handle the throughput.
In addition we also log a small amount of environmental data every 60s in the analog data logs.

The serial log starts with a shorter heater followed by the 'static' data, then we send the longer header for the dynamic data section.
This header consists of 444 ASCII characters. To avoid overflow in either the output buffer of my system or the input buffer of the OLA we are chunking this in 5 blocks of roughly equal size, send 1s apart.
The dynamic data output is shorter, but also sent in chunks to avoid overflow. It records ok in the log.
We also want the timestamp added to the data lines and trigger that with \r after each data block is complete.

What we are seeing is that the header is missing a chunk of characters from the header section in the log.
E.g. the first 87 characters are logged, then there are 50 characters missing, then the remainder of the header is logged again.
It seems that without a timestamp the header logs completely, so this could be an artefact of the timestamp insertion.

We have slowed down the timing of the individual chunks of the header section without changing the outcome. 
We also reduced the amount of characters per chunk by abbreviating headers, but the log is still missing data in the second chunk sent out.

For now I want to understand what is happening here and I am looking for a workaround to circumvent this issue.
How can we send out the header characters and have all of them logged? I am happy to change timing and chunking to suit the capabilities of the OLA, but this has been a trial and error approach up to this point without getting a result.

Of course if this is an easy fix in the OLA firmware then please take this as a bug report.

Regards, Peter

Beepnblink, 2025-02-03:
Update:

I thought that I was already feeding the OLA manageable serial data chunks slowly.
But
Slowing it down to a chunk (~70 characters) every 5 seconds got finally all the data logged.
That was WAY slower than I anticipated. 
For future reference, is there any way to recognise an internal data buffer overflow, e.g. is there a debug option that would indicate what is happening?

Regards, Peter

PaulZC, 2025-02-03:
Hi Peter (@Beepnblink ),

Which version of the firmware are you using? 2.9 or 2.6?

Regardless, please try the new v2.10 BETA firmware:

https://github.com/sparkfun/OpenLog_Artemis/blob/main/Binaries/OpenLog_Artemis-V10-v210_BETA.bin

I've restructured the serial logging code in ```loop()``` to make it easier to follow - and hopefully make it more reliable. I couldn't see an obvious explanation for your issue, but the original code wasn't well structured and there may have been a gotcha hiding in there. For best results, please reset the firmware settings with option "r" and start afresh.

Let me know how it goes.

Best wishes,
Paul

Beepnblink, 2025-02-04:
Hi Paul,

I was using the V2.9 before and loaded your 2.10 BETA now.
Speeding up the data chunks for the serial port to 1/s works and serial data is logged ok. I did not try it faster or increase the chunk size as this is all I need here.
Hoever I can't get the 'Log to Micro SD ' setting to stick (terminal output / 1). It shows enabled when set, but on next bootup the setting is back to disabled. No sensor data logs appear on the SD card.
Not sure what is causing this, or if this depends on another setting?

Regards, Peter

PaulZC, 2025-02-04:
Hi Peter,

Progress - good! I'm glad the serial logging is working better.

I haven't made any changes to 'Log to Micro SD'. That should be as it was in previous versions. Please try again and make sure you fully exit the menus before you disconnect the power. The menu settings are only saved when you fully exit the menus and return to logging.

Best wishes,
Paul

closed 2025-02-06 by PaulZC