sparkfun/OpenLog_Artemis #209: Edit Log file name

ndeschamps-isto, 2025-02-05:
Hi OLA community,

Is there a way to configure the filenames dataLog0000.txt and serialLog0000.txt? I am planning to use 5 OLA devices for field measurements at 5 different stations. It would be useful to add the station name to the filename, so I can easily identify which log corresponds to which station. This would also make it easier to import and process my logs using an R or Python script.

Using (DEV-16832) without IMU; Firmware V10-v28.

Thanks for your help !

PaulZC, 2025-02-05:
Hi @ndeschamps-isto ,

Sadly, no, that's not possible with the current firmware. The ```dataLog``` and ```serialLog``` filename prefixes are fixed...

I will leave this open and change it into a "Feature Request". We will try to implement it if/when time permits. If you have the time to add this feature, please do and send us a Pull Request.

Best wishes,
Paul

ndeschamps-isto, 2025-02-05:
Thanks for your quick answer. That was my expectation. I did not try to update the code myself, as I'm not familiar with the whole project and I'm not a proper C++ developer. However, if I have time and feel able to do it, I will try.

I'm sure this update would be appreciated by many users! Thanks for considering it.

Best wishes.