sparkfun/OpenLog_Artemis #210: Serial log does not record the last measurements

ndeschamps-isto, 2025-02-05:
Hi OLA community,

I’m planning to use OLA for water level measurements with the ultrasonic sensor [MB7389](https://maxbotix.com/products/mb7389?srsltid=AfmBOoqvblQqk6Abapth8eKVYTom9hnU9MDeyE5IZl42BAos4a21U1A-). It will be deployed in the field for long-term use (years) and will log data at a 30-minute interval. We plan to change the battery and read the SD card periodically (every 2-3 months). 

My sensor can communicate in analog (0-3.3V). Despite the OLA's 0-2V range, this works fine for my needs, as I don't require the full range. However, I get better precision using the serial (TTL) output, so I’d like to take advantage of that.

The sensor's output is an ASCII capital “R”, followed by four ASCII digits representing the range in millimeters, followed by a carriage return (ASCII 13). The serial data format is 9600 baud, 8 data bits, no parity, and one stop bit (9600-8-N-1). The response time is around 180ms.

Here are the main settings I’ve changed from the default:

- **1: Configure Terminal Output**
  - 1: Log to microSD: *Enabled*
  - 2: Log to Terminal: *Disabled*
  - 5: Set Log Rate in seconds between readings: *5*
  - 7: Output Actual Hertz: *Disabled*
  - 13: Output Column Titles: *Disabled*
  - 21: Minimum awake time between sleeps: *200ms*
- **2: Configure Time Stamp**
  - 1: Log Date: *Enabled*
  - 2: Log Time: *Enabled*
- **5: Configure Analog Logging**
  - 1: Log analog pin 11 (2V Max): *Enabled* (Triggering is disabled)
  - 5: Log output type: *Raw ADC reading*
  - 6: Log VIN (battery) voltage (6V Max): *Enabled*
- **4: Configure Serial Logging**
  - 1: Log serial data: *Enabled* (analog logging on RX/A13 pin disabled)
  - 2: Add RTC timestamp when token is received: *Enabled*
  - 3: Timestamp token: *13 (Decimal) = Carriage Return*
- **7: Configure Power Options**
  - 1: Turn off Qwiic bus power between readings (>2s): *Yes*
  - 7: Serial Tx and Rx pins during sleep: *Disabled*

I’ve tried various settings, different FAT32 SD cards, and ensured I disconnect the power before removing the SD card, but I always encounter the same issue: I miss the last 2-5 measurements in the serial log. This happens regardless of the log rate. 

Additionally, I’m seeing some weird output serial format and a few minor errors (though these are not a major issue, as I can write a post-processing script to clean the logs).

datLog
```
09/08/2024,30:01:27.86,3020,5.45,
09/08/2024,30:01:32.86,2982,5.46,
09/08/2024,30:01:37.86,1557,5.45,
09/08/2024,30:01:42.86,1575,5.45,
```

serialLog
```
^09/08/2024,30:01:17.86
R0589
^09/08/2024,30:01:22.86
R0570
^09/08/2024,30:01:22.86
R0570
^09/08/2024,30:01:22.86
R0569
^09/08/2024,30:01:27.86
^ 9
^09/08/2024,30:01:27.86
R0569
^09/08/2024,30:01:27.86
R0569
^09/08/2024,30:01:32.8
```

Don’t pay attention to the datetime (trouble with `__date__` and `__time__` compiler macros... another issue), but you can see that my last dataLog is at `30:01:42` and my last serialLog is at `30:01:27` (so I miss the last 3 measurements). Furthermore, the serial data is logged 2 or 3 times, and the last row is incomplete (with no value).

I’m using the OLA (DEV-16832) without the IMU, and Firmware V10-v28.

I hope someone can help me ! 
Thank you.

PaulZC, 2025-02-05:
Hi @ndeschamps-isto ,

Can you please try the new v2.10 BETA binary:

https://github.com/sparkfun/OpenLog_Artemis/blob/main/Binaries/OpenLog_Artemis-V10-v210_BETA.bin

It includes some changes to the serial logging code - see issue #208 for more details.

Thank you,
Paul

ndeschamps-isto, 2025-02-06:
Hi Paul,

Thanks for your help. I thought I was using the latest firmware but didn't read issue #208 carefully enough.
v2.10 works better! The serial log is cleaner, but I still miss the very last measurement.

It seems acceptable for some field tests, and I will keep an eye on the next updates.

I really like the OLA project. If it proves reliable in the field, it could be a great, low-cost, open-source and user-friendly logger (especially for people who aren't familiar with Arduino/C++ coding) for various environmental monitoring applications. 

Best regards.
Nicolas.

PaulZC, 2025-02-06:
Hi Nicolas,

Good. I am glad v2.10 is working better for you. I will do a full release soon.

There is a 500ms idle timer in the serial logging code. Data is written to the serialLog file when: the incomingBuffer is full (512 bytes); or the 500ms timer expires. Data received in the previous 500ms will be lost when the OLA is powered off, or goes to sleep...

If the power has been removed, the priority is to close the files as quickly as possible. I think it is acceptable that data is lost in this case.

But when going to sleep, it also does the same thing. And it should not, because there is time to write the data to file. I should change the code so that it updates the serialLog file before closing it.

I will add this to v2.10.

Best wishes,
Paul

closed 2025-02-06 by PaulZC