sparkfun/OpenLog_Artemis #211: [Feature Request] Allow del* or rm* to delete all log files

knikolaus-pice, 2025-02-06:
Using sz* to download all the available log files is great. Having a similar feature to clear the microSD card of those files would be helpful. I am using the Artemis Openlog as a ice core borehole logging tool and what I would like to do is download all the log files and then wipe the memory of the microSD card (other than the configuration files) for the next run.

PaulZC, 2025-02-06:
Hi @knikolaus-pice ,

Sure. This should be ~easy to add... (Famous last words!)

Best wishes,
Paul

closed 2025-02-06 by PaulZC

PaulZC, 2025-02-06:
I have added this feature in [v2.10](https://github.com/sparkfun/OpenLog_Artemis/releases/tag/v2.10). The updated binary will be available in a few minutes.

commit c3858a78 (2025-02-06): Add DEL * / RM * to resolve #211