sparkfun/OpenLog_Artemis #214: No Magnetometer values for test sketch OLA_IMU_Basic_Standadlone.ino

starrobin34, 2025-04-23:
Hello Sparkfun Team, 

with the OLA_IMU_Basic_Standadlone.ino everything seems to be working fine, except I am not getting any values for the Magnetometer.  I haven't changed anything but I've tried examples from ICM_20948.h and I am getting the same result. My board is the OLA V1 (redboard) with IMU. Set up in Arduino IDE as "RedBoard Artemis ATP" 

IMU ouput: 
Scaled. Acc (mg) [  00181.64, -00090.82, -00979.98 ], Gyr (DPS) [ -00000.33, -00000.66, -00000.42 ], Mag (uT) [  00000.00,  00000.00,  00000.00 ], Tmp (C) [  00030.14 ]

Appreciate help, let me know if you need further information.

starrobin34, 2025-04-23:
https://github.com/wollewald/ICM20948_WE/issues/7 seems to be the same issue

starrobin34, 2025-04-23:
Hey Paul, 

all good I can work with the old version! Thanks for the quick reply 
Cheers Robin

PaulZC, 2025-04-23:
Hi Robin (@starrobin34 ),

Also, please check that you do not have the DMP enabled in the 20948 library when running the OLA_IMU_Basics sketch. The magnetometer needs to be configured differently when using the DMP, so the IMU ```begin``` terminates early leaving it to other code to finish configuring the mag. Maybe you had the DMP enabled in a recent version of the 20948 library, and regressing to 1.2.8 disabled it?

I hope this helps,
Paul