sparkfun/OpenLog_Artemis #216: Feature request: Add support for VEML6030 ambient light sensor

flexpeteps, 2025-06-28:
The addition of the VEML7700 ambient light sensor was a very useful addition, but this sensor is unfortunately no longer available. VEML6030 is the suggested replacement. It would be great if support for it could be added, which I hope is relatively straightforward given the support for VEML7700. Thank you!

PaulZC, 2025-08-07:
Hi @flexpeteps ,

Apologies for the slow reply.

The VEML7700 is available again:

https://www.sparkfun.com/sparkfun-ambient-light-sensor-veml7700-qwiic.html

Best wishes,
Paul

PaulZC, 2025-08-07:
Note: it looks like adding the VEML6030 would be possible - but only on address 0x48, with the ADDR pin pulled high.

With the ADDR pin low, the address is 0x10 and the device will - I think - appear identical to the VEML7700. The register map looks identical.