sparkfun/OpenLog_Artemis #219: Onboard ICM-20948 IMU fails to initialize (multiple OpenLog Artemis IMU units, v2.10 firmware)

nairck, 2025-08-07:
**Board**: SparkFun OpenLog Artemis 16832 (with IMU onboard)  
**Firmware**: v2.10 (official release)  
**Power source**: USB 5V (also tried battery)
**microSD card**: 64GB Class 10 Raspberry Pi SC1629
**Tested on**: 4 separate boards with onboard IMU  
**Mfg. Lot Num.**: 174730 (all the same lot)
**Vendor**: Digikey
**Platform**: Arduino IDE 1.8.19, Artemis Firmware Uploader v3.1.0

---

### Issue:
The onboard ICM-20948 IMU fails to initialize on **multiple new boards**. Each unit outputs the following via serial monitor on startup:

```
⸮)⸮⸮ցV⸮⸮9{l=⸮⸮⸮jn⸮⸮⸮Artemis OpenLog v2.10
Logging to: dataLog00030.TXT
Finding the next available log file.
This could take a long time if the SD card contains many existing log files.
Logging to: serialLog00000.TXT
ICM-20948 failed to init.
SD card online
Data logging online
Serial logging online
IMU offline - or not present
Logging to microSD card with no terminal output
Identifying Qwiic Muxes...
Identifying Qwiic Devices...
Autodetect complete
Content-Type: text/csv

rtcDate,rtcTime,
```

The i2c interfacing with eternal sensors, SD, and serial logging work fine otherwise.
No onboard IMU activity, no successful init on any board. 
Tried the OLA_IMU_Basics sketch and only see "Wrong ID" show up trying to find the IMU. Trying i2c version of this sketch I see "stack underflow"
Tried V2.4 out of box, and updated bootloader and firmware to v2.10 with no difference.
This occurs even after firmware reflash via Artemis Firmware Uploader v3.1.0  

I'm trying to see if Digikey has a different lot they can replace my OLAs with.

PaulZC, 2025-08-07:
HI Adam,

OK - thanks! That's very strange...

Please have a good look at the surrounding components. Are any of those missing? (Apart from the missing resistor next to the "G" of "AGX". That one is supposed to be missing.)

[image][image]

Best wishes,
Paul

nairck, 2025-08-07:
Hey Paul,

Looking at each board, they all appear fully populated just like they are supposed to be, with only a single resistor missing under the G as you
Indicate. 

Adam

[image]

PaulZC, 2025-08-07:
NIce photo - thank you!

Please try popping out the SD card, and reset. Is the IMU still not detected?

Are the four faulty boards the only four boards you have? Do you have other boards which work OK?

nairck, 2025-08-07:
I noticed in notes somewhere 'older FAT16 microSD cards up to 32GB are supported', but I'm using the 64GB Raspberry Pi Class 10 microSD card (SC1629). By default, I see these are configured at exFAT - is this maybe causing issues?

PaulZC, 2025-08-07:
Thanks Adam,

There's nothing 'wrong' (per se) with your 64GB cards. exFAT is supported too - thanks to a newer version of SdFat.

The issue will be the amount of current your cards are taking on power-up ("in-rush"). I suspect they're taking a big gulp of current, big enough to cause the voltage rail to dip while the IMU is being started. It's a [known issue](https://community.sparkfun.com/t/openlog-artemis-wont-power-on-with-microsd-card-inserted/44531).

Hand on heart, we should have included more decoupling on the OLA around the SD card power switch (FET). But, we were testing with small cards and didn't see this issue at the time.

Apologies for the hassle,
Paul

PaulZC, 2025-08-07:
> Can multiple delays be added to slow down the start up sequence

Yes, but you'd need to recompile the firmware to add those... Please see: [COMPILE_BINARY.md](https://github.com/sparkfun/OpenLog_Artemis/blob/main/COMPILE_BINARY.md)

nairck, 2025-08-07:
I did try compiling the arduino sketch but had many compile errors, after downloading the libraries. I’ll try to follow those instructions more carefully.

Would this be something I could request for the next patch?

nairck, 2025-08-08:
Thank you for the help and guidance, and speedy replies, I'm very pleased with the support :)

I am going away for the weekend but will be back in a few days (Monday) to try adjusting the code. I'll make sure to follow the detailed instructions this time hah..

For logging, I am trying to capture 50Hz Accel/Gyro sampling onboard and an additional daisy chained 6dof accel/gyro plus a BME280 for temp/pressure/humidity. I sampled this rate with this amount of data over 2 days and the text file reached about 1GB. The aim for my logger setup is to reach 4-6 weeks, so this is why I really wanted 64GB or more. My small 2 day test suggests I should about 4 weeks of logging on 64GB.

nairck, 2025-08-09:
Update - I can confirm the 32GB version of the Raspberry Pi microSD card works - the IMU is initialized properly and everything else works. I can also confirm a Kingston 32GB microSD works with the IMU as expected.

I have ordered more 32 and 64 GB Kingston brand microSD’s for testing, hoping I can find a 64 that works

nairck, 2025-08-11:
Okay so I can also confirm that the Kingston SDCS3/64GB and Kingston SDCE/32GB work without trouble (IMU online and everything else functional), so I found a 64GB microSD that works!

I may put the code adjustments on the back burner as I now have working loggers and the 64GB raspberry pi cards can be used fro something else.

Thank you for working through this with me!